carla-simulator/carla #2417: Build 'package' target failed during BuildCookRun.  Directory delete failed.

sound-design, 2020-01-30:
I finally have fought my way through the process of building Carla 0.9.7 on Ubuntu 18.04 LTS where I am now trying to `make` the last build target: `make package`.

The process grinds on for about an hour and then fails at the `BuildCookRun` command line with the following error dump:
```
Running AutomationTool...

Fixing inconsistent case in filenames.
Setting up Mono
Compiling AutomationTool with xbuild
xbuild Source/Programs/AutomationTool/AutomationTool.csproj /p:Configuration=Development /p:Platform=AnyCPU /verbosity:quiet /nologo /p:TargetFrameworkProfile=
Compilation Succeeded
Start UAT: mono AutomationTool.exe BuildCookRun -project=/home/user1/Documents/GitHub/ME/carla/Unreal/CarlaUE4/CarlaUE4.uproject -nocompileeditor -nop4 -cook -stage -archive -package -clientconfig=Shipping -ue4exe=UE4Editor -prereqs -targetplatform=Linux -build -utf8output -archivedirectory=/home/user1/Documents/GitHub/ME/carla/Dist/CARLA_Shipping_0.9.7-12-g1f7669e4
Parsing command line: BuildCookRun -project=/home/user1/Documents/GitHub/ME/carla/Unreal/CarlaUE4/CarlaUE4.uproject -nocompileeditor -nop4 -cook -stage -archive -package -clientconfig=Shipping -ue4exe=UE4Editor -prereqs -targetplatform=Linux -build -utf8output -archivedirectory=/home/user1/Documents/GitHub/ME/carla/Dist/CARLA_Shipping_0.9.7-12-g1f7669e4 -compile
WARNING: Failed to delete directory /media/user1/Stn3-Linux-BionicBU/OVERFLOW/UnrealEngine_4.22/Engine/Binaries/DotNET/AutomationScripts/AllDesktop in 10 attempts.
WARNING: Exception in mscorlib: Directory /media/user1/Stn3-Linux-BionicBU/OVERFLOW/UnrealEngine_4.22/Engine/Binaries/DotNET/AutomationScripts/AllDesktop is not empty
         Stacktrace:   at System.IO.Directory.Delete (System.String path) [0x0006f] in <98fac219bd4e453693d76fda7bd96ab0>:0 
           at System.IO.Directory.RecursiveDelete (System.String path) [0x00051] in <98fac219bd4e453693d76fda7bd96ab0>:0 
           at System.IO.Directory.Delete (System.String path, System.Boolean recursive) [0x00009] in <98fac219bd4e453693d76fda7bd96ab0>:0 
           at AutomationTool.InternalUtils.SafeDeleteEmptyDirectory (System.String Path, System.Boolean bQuiet) [0x0003d] in <af672cfb7c4e4983888d4b531dd21833>:0
ERROR: Failed to delete directory '/media/user1/Stn3-Linux-BionicBU/OVERFLOW/UnrealEngine_4.22/Engine/Binaries/DotNET/AutomationScripts'
       (see /home/user1/Library/Logs/Unreal Engine/LocalBuildLogs/Log.txt for full exception trace)
AutomationTool exiting with ExitCode=1 (Error_Unknown)
RunUAT ERROR: AutomationTool was unable to run successfully.
Util/BuildTools/Linux.mk:16: recipe for target 'package' failed
make: *** [package] Error 1
```
One odd thing I noticed is that after this error occurs and I am back at the shell prompt, `gzip` is still running in the background.  After writing about 2 to 3 GB to disk, consuming nearly an entire core or two by itself on the CPI, it finally ends.  

How can I fix this problem and proceed?  Is the error actually because the `make package` target, as one of its steps, does a large gzip operation and perhaps while that operation is in progress, the AutomationScripts directory can not be deleted, resulting in the error message shown above?  If not, what is the problem?  I'd really like to get this fixed quickly.

jeffreykxiao, 2020-01-31:
The AutomationScripts directory being deleted is a problem I've encountered with UE4 in the past, but I can't remember exactly how I encounter it.. I think the usual case would be something like using binary editor from Epic and not using a `-nocompile` flag. The UAT script actively deletes the directory if trying to compile. Frankly, I don't really recall any more details around that, but you could always try to change what arguments are getting passed to `BuildCookRun`

Aside from that, though, is there a particular reason you want to `make package`? Personally, I just use the editor through `make launch`. I've used `make package` without issues before, but I always pass "ARGS=mapnames" since I just build maps for import into official releases..

Sorry I don't have exact details on your issue, I can try to reproduce it later, but hopefully this gives you at least a little bit to go on.

roschler, 2020-01-31:
@jeffreykxiao - I ran the make again and this time it went through.  I'm still wondering if it was due to the build script trying to delete that directory when some previous build strep initiated a long running gzip operation, that was not executed with a wait-for-completion status by the initiating shell process.

re: make package

I'm just trying to get this thing working.  I'm having a truly brutal experience getting this to work.  It took 3 to 4 days just to get to the point I could make all the targets.  Now when I run `manual_control.py` I'm getting the currently unresolved issue with an error regarding `PyString_type` being an undefined symbol.  My web searching led me to this Stack Overflow post that claims it's a case of a Python 2 built file creeping in and that error manifesting when Python 3 tries to run.  In the context of that post the offending library is the libboost library.

https://stackoverflow.com/questions/25188861/libboost-python3-so-1-56-0-undefined-symbol-pyclass-type

https://github.com/carla-simulator/carla/issues/2256

So as I said, I'm trying to build the `make package` target in yet another desperate attempt to get this to work.

sound-design, 2020-01-31:
@jeffreykxiao - I ran the make again and this time it went through. I'm still wondering if it was due to the build script trying to delete that directory when some previous build strep initiated a long running gzip operation, that was not executed with a wait-for-completion status by the initiating shell process.

re: make package

I'm just trying to get this thing working. I'm having a truly brutal experience getting this to work. It took 3 to 4 days just to get to the point I could make all the targets. Now when I run manual_control.py I'm getting the currently unresolved issue with an error regarding PyString_type being an undefined symbol. My web searching led me to this Stack Overflow post that claims it's a case of a Python 2 built file creeping in and that error manifesting when Python 3 tries to run. In the context of that post the offending library is the libboost library.

https://stackoverflow.com/questions/25188861/libboost-python3-so-1-56-0-undefined-symbol-pyclass-type

#2256

So as I said, I'm trying to build the make package target in yet another desperate attempt to get this to work.

closed 2020-04-07 by stale[bot]