carla-simulator/carla #2420: New Vehicle not affected by physics

Georgini82, 2020-01-31:
I followed the instructions to add a new vehicle to carla from https://carla.readthedocs.io/en/latest/how_to_add_assets/ . 
But for some reason when I spawn the car trough the PythonAPI it just stays in the air instead of falling to the road like the other cars.

Did I missed something? It looks like the vehicle is spawn as a static object that is not affected by the physics asset.

[image]

Georgini82, 2020-02-17:
Hello, I realised I made a mistake. 
So for future references: the body of the car cannot have simulation=kinematic. 
It was that parameter that keep the car locked in the place where it spawn.

Regards

closed 2020-02-17 by Georgini82

iasonastsaousis, 2025-02-17:
Where is that parameter found?