carla-simulator/carla #2428: Any way to speed up synchronous mode with no rendering? (0.9.6)

JamesKCS, 2020-01-31:
Thank you for the outstanding work on this simulator; it is very useful for my research needs!

I understand synchronous mode will be slower than async mode.  As the docs say, 

> Secondly, the synchronous mode imposes a significant performance penalty. There is a price in waiting for the render thread to have the images ready and halting the simulation when the client is slow. There is a trade-off in using the synchronous mode.

However, I am hoping to be able to use synchronous mode for consistency for a reinforcement learning problem.  Based on the above quote, I expected synchronous performance to increase drastically when I turned rendering off.  I was getting about .04 seconds per `world.tick()` call before turning graphics off.  To my surprise, turning graphics off had (`settings.no_rendering_mode = True` and/or launching with opengl) had no significant effect on this interval, it still takes about .04 seconds per `world.tick()` call.  Is this expected and is there any way around this?  Or am I perhaps making a trivial error?

**More details:**
Running CARLA 0.9.6
The relevant code looks something like this:
```
make_fixed_time_step(world)
make_synchronous(world)
set_no_rendering_mode(world)

def make_fixed_time_step(world, time_step = .05):
    settings = world.get_settings()
    settings.fixed_delta_seconds = time_step
    world.apply_settings(settings)

def set_no_rendering_mode(world):
    settings = world.get_settings()
    settings.no_rendering_mode = True
    world.apply_settings(settings)

def make_synchronous(world):
    settings = world.get_settings()
    settings.synchronous_mode = True
    world.apply_settings(settings)
```

Thank you in advance!

jerryangit, 2020-02-06:
I'm having the same issue, I need to use synchronous mode to ensure consistent results but the significant slowdown is quite an annoyance. This other issue seems to be about the same problem: https://github.com/carla-simulator/carla/issues/1771#issue-457886448.

JamesKCS, 2020-02-14:
Yeah, I think the issue you linked is simply asking why synchronous mode is slower.  It's unfortunate, but I think the answer to that issue is that synchronous mode will always be significantly slower because of how CARLA is put together.  However, based on the docs (and the answers in the issue you linked), I'd still expect a significant speed-up for turning off rendering.

The fact that it always gives about 20-25 fps, and that turning rendering results in no speedup (according to the docs, rendering is a big part of why synchronous mode is so much slower), makes me think that either 1) I'm making some sort of trivial error and rendering is still occurring or 2) synchronous mode is being artificially slowed-down somehow. Whichever thing is occuring, hopefully some sort of solution can be suggested by the CARLA team.  There is no way the physics of 4 cars driving, with no rendering, around should be limiting it to 20-25 fps on a high-end machine, even in synchronous mode...

cbrewitt, 2020-02-25:
I've been having the same problem. I'm getting 23 FPS with rendering disabled and only one vehicle.

closed 2020-05-02 by stale[bot]

Kait0, 2024-03-06:
The question was never answered and this is still an issue in the latest version 0.9.15.