carla-simulator/carla #2441: How to load a custom map

ndeshp2s, 2020-02-04:
Is there a way to load a custom made town from config.py file for example? Or by running "CarlaUE4.sh" ?

I have created a custom environment and it works fine from unreal engine (after pressing the play button). How I want to register this environment so that I can load it from code directly.

Is there a way to do this?

a-jahani, 2020-02-05:
You need to register the map in the cook packages secotion in 'project settings' inside unreal engine. 
Then you can package it.
You might also need a  .xodr file for it .you can use an empty one(you will see the error saying this file is missing)

a-jahani, 2020-02-05:
After packaging (make package) client.get_available_maps() method should list the custom map.

closed 2020-04-12 by stale[bot]