carla-simulator/carla #2443: RuntimeError in linux

EgorFollia, 2020-02-05:
I have installed Carla using this doc:
> https://carla.readthedocs.io/en/latest/how_to_build_on_linux/

Now I am trying to run a python script from /PythonApi/examples and I got this:
```
user@user:/home/user/carla/PythonAPI/examples$ ./automatic_control.py 
pygame 1.9.6
Hello from the pygame community. https://www.pygame.org/contribute.html
INFO: listening to server 127.0.0.1:2000

    Example of automatic vehicle control from client side.

Traceback (most recent call last):
  File "./automatic_control.py", line 685, in <module>
    main()
  File "./automatic_control.py", line 678, in main
    game_loop(args)
  File "./automatic_control.py", line 593, in game_loop
    world = World(client.get_world(), hud, args.filter)
RuntimeError: time-out of 5000ms while waiting for the simulator, make sure the simulator is ready and connected to 127.0.0.1:2000
```
How to resolve this problem?

ndeshp2s, 2020-02-05:
@EgorFollia You need to run the server first. Without that automatic_control.py (which is a client) would fail.
Assuming you have installed carla from source you follow the following steps:
- In one terminal run "./CarlaUE4.sh" (you will find CarlaUE4.sh in Dist/CARLA_Shipping_.../LinuxNoEditor/). You can refer to [https://github.com/carla-simulator/carla/issues/2032](url)
- Then in another terminal run "automatic_control.py".

EgorFollia, 2020-02-05:
> @EgorFollia You need to run the server first. Without that automatic_control.py (which is a client) would fail.
> Assuming you have installed carla from source you follow the following steps:
>
> * In [...]

Hm, but the command:
> user@user:/home/user/carla/Dist/CARLA_Shipping_0.9.7-30-g5a904579$ ls
- didn't return anything. Why does this folder might be empty?

EgorFollia, 2020-02-09:
@ndeshp2s But this folder is empty:
`user@user:/home/user/carla/Dist$ ls CARLA_Shipping_0.9.7-30-g5a904579/` - returns nothing

closed 2020-02-12 by EgorFollia