carla-simulator/carla #2449: build failure

jaypee92, 2020-02-07:
Makefile:290: recipe for target 'CarlaUE4Editor' failed

$ make launch
Setup.sh: llvm-7.0 already installed.
Setup.sh: boost-1.69.0-c7 already installed.
Setup.sh: rpclib-v2.2.1_c2-c7 already installed.
Setup.sh: gtest-1.8.1-c7 already installed.
Setup.sh: recast-cdce4e-c7 already installed.
Setup.sh: CARLA version 0.9.7-43-g1449b1d.
Setup.sh: Generating CMake configuration files.
Setup.sh: Success!
BuildLibCarla.sh: Building LibCarla "Server.Release" configuration.
ninja: no work to do.
[1/1] Install the project...
-- Install configuration: "Server"
BuildLibCarla.sh: Success!
BuildCarlaUE4.sh: Using Unreal Engine at '/home/jaypee/UnrealEngine_4.24.2'
BuildCarlaUE4.sh: Build CarlaUE4 project.
make[1]: Entering directory '/home/jaypee/carla/Unreal/CarlaUE4'
bash "/home/jaypee/UnrealEngine_4.24.2/Engine/Build/BatchFiles/Linux/RunMono.sh" "/home/jaypee/UnrealEngine_4.24.2/Engine/Binaries/DotNET/UnrealBuildTool.exe" CarlaUE4Editor Linux Development  -project="/home/jaypee/carla/Unreal/CarlaUE4/CarlaUE4.uproject" 

Running Mono...

Fixing inconsistent case in filenames.
Setting up Mono
~/UnrealEngine_4.24.2/Engine ~/carla/Unreal/CarlaUE4
Using 'git status' to determine working set for adaptive non-unity build (/home/jaypee/UnrealEngine_4.24.2).
Creating makefile for CarlaUE4Editor (no existing makefile)
Parsing headers for CarlaUE4Editor
  Running UnrealHeaderTool "/home/jaypee/carla/Unreal/CarlaUE4/CarlaUE4.uproject" "/home/jaypee/carla/Unreal/CarlaUE4/Intermediate/Build/Linux/B4D820EA/CarlaUE4Editor/Development/CarlaUE4Editor.uhtmanifest" -LogCmds="loginit warning, logexit warning, logdatabase error" -Unattended -WarningsAsErrors
/home/jaypee/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Actor/ActorAttribute.h(29) : LogCompile: Error: 'EActorAttributeType' does not have a 0 entry! (This is a problem when the enum is initalized by default)
/home/jaypee/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Vehicle/VehicleInputPriority.h(24) : LogCompile: Error: 'EVehicleInputPriority' does not have a 0 entry! (This is a problem when the enum is initalized by default)
/home/jaypee/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Util/ActorAttacher.h(27) : LogCompile: Error: 'EAttachmentType' does not have a 0 entry! (This is a problem when the enum is initalized by default)
/home/jaypee/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Settings/QualityLevelUE.h(27) : LogCompile: Error: 'EQualityLevel' does not have a 0 entry! (This is a problem when the enum is initalized by default)
/home/jaypee/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Traffic/TrafficSignBase.h(31) : LogCompile: Error: 'ETrafficSignState' does not have a 0 entry! (This is a problem when the enum is initalized by default)
Makefile:290: recipe for target 'CarlaUE4Editor' failed
make[1]: *** [CarlaUE4Editor] Error 5
make[1]: Leaving directory '/home/jaypee/carla/Unreal/CarlaUE4'
Util/BuildTools/Linux.mk:7: recipe for target 'launch' failed
make: *** [launch] Error 2

carla version : 0.9.6
Unreal Engine 4.24.2

roser-br, 2020-02-12:
Hi! 
I'm facing the same error when trying to build Carla 0.9.7 on Ubuntu 18.04 and Unreal Engine 4.22.3. 
Does someone know what could be to done to solve this? 
Thanks in advance!

Roosstef, 2020-02-12:
I think this error so related to the libpng16 Installation. I faced the Same Issue couple Weeks ago. Installing libpng16 and libpng16-dev fixed it in My case.

roser-br, 2020-02-18:
Thanks for your help @Roosstef! However, it didn't work for me...

stevelaclasse, 2020-02-24:
did you try to install clang and cmake before building carla ?

closed 2020-05-02 by stale[bot]