carla-simulator/carla #2456: [0.9.7] Loosing connection to remote Carla instance

semmelknoedel, 2020-02-10:
I'm running a docker container of version 0.9.7 on a remote machine. When I start manual_control.py there it works fine.
When I start manual_control.py on my local machine I get the following output:

 ```
/usr/bin/python3.6 /path/to/carla/PythonAPI/examples/manual_control.py --host x.x.x.x
pygame 1.9.6
Hello from the pygame community. https://www.pygame.org/contribute.html

Welcome to CARLA manual control.

Use ARROWS or WASD keys for control.

    W            : throttle
    S            : brake
    AD           : steer
    Q            : toggle reverse
    Space        : hand-brake
    P            : toggle autopilot
    M            : toggle manual transmission
    ,/.          : gear up/down

    TAB          : change sensor position
    `            : next sensor
    [1-9]        : change to sensor [1-9]
    G            : toggle radar visualization
    C            : change weather (Shift+C reverse)
    Backspace    : change vehicle

    R            : toggle recording images to disk

    CTRL + R     : toggle recording of simulation (replacing any previous)
    CTRL + P     : start replaying last recorded simulation
    CTRL + +     : increments the start time of the replay by 1 second (+SHIFT = 10 seconds)
    CTRL + -     : decrements the start time of the replay by 1 second (+SHIFT = 10 seconds)

    F1           : toggle HUD
    H/?          : toggle help
    ESC          : quit

INFO: listening to server x.x.x.x:2000
WARNING: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API 
WARNING: Client API version     = 0.9.7-41-g68d7418e 
WARNING: Simulator API version  = 0.9.7 
No recommended values for 'speed' attribute
Traceback (most recent call last):
  File "/path/to/carla/PythonAPI/examples/manual_control.py", line 1034, in <module>
    main()
  File "/path/to/carla/PythonAPI/examples/manual_control.py", line 1026, in main
    game_loop(args)
  File "/path/to/carla/PythonAPI/examples/manual_control.py", line 943, in game_loop
    controller = KeyboardControl(world, args.autopilot)
  File "/path/to/carla/PythonAPI/examples/manual_control.py", line 276, in __init__
    world.player.set_autopilot(self._autopilot_enabled)
RuntimeError: time-out of 5000ms while waiting for the simulator, make sure the simulator is ready and connected to localhost:2000
Fatal Python error: PyEval_SaveThread: NULL tstate

Process finished with exit code 134 (interrupted by signal 6: SIGABRT)

 ```

I have the same problem with spawn_npc.py. But it actually spawns some npcs at the remote machine before the connection crashes (so at the beginning there is a connection). Also I am confused that the last error message is `localhost:2000` but I clearly gave an address.

Both, remote and locale machine, are running with Ubuntu 18.04 LTS.

Has someone seen this problem before?

pawel-ziecina, 2020-02-11:
I'm observing similar issue on local machine. Steps to reproduce:

1. Start original carla 0.9.7.3 release with

```
./CarlaUE4.sh -opengl -carla-rpc-port=36540 -carla-streaming-port=0 -quality-level=Low -nosound
```

2. Check if carla server properly binded

```
netstat -nlp|grep -i carla

tcp        0      0 0.0.0.0:44581           0.0.0.0:*               LISTEN      21795/CarlaUE4-Linu 
tcp        0      0 0.0.0.0:36540           0.0.0.0:*               LISTEN      21795/CarlaUE4-Linu 
```

3. Run example `spawn_npc.py` script:

```
PYTHONPATH=$PWD/PythonAPI/carla/dist/carla-0.9.7-py3.5-linux-x86_64.egg python3.6 ./PythonAPI/examples/spawn_npc.py -p 36540 --host 127.0.0.1
```

Observing:

```
destroying 0 vehicles

destroying 0 walkers

done.
Traceback (most recent call last):
  File "./PythonAPI/examples/spawn_npc.py", line 226, in <module>
    main()
  File "./PythonAPI/examples/spawn_npc.py", line 124, in main
    for response in client.apply_batch_sync(batch):
RuntimeError: time-out of 5000ms while waiting for the simulator, make sure the simulator is ready and connected to localhost:2000
Fatal Python error: PyEval_SaveThread: NULL tstate

```

semmelknoedel, 2020-02-12:
As a temporal solution I'm using port forwarding right now:
`ssh -L 2000:localhost:2000 -L 2001:localhost:2001 -L 2002:localhost:2002 user@remote_host`

jackbart94, 2020-02-13:
Hi,

A solution to this issue will be in master very soon, thanks for pointing it out!

liespace, 2020-02-27:
@jackbart94 Has this bug been fixed? I use the latest version docker but still meet this bug.

jackbart94, 2020-02-27:
@liespace It's being merged in master as we speak!

closed 2020-05-06 by stale[bot]