carla-simulator/carla #2458: Control Wheel speed

LuccaFaro, 2020-02-11:
Hello everyone!

Sorry if my question is silly or doesn't make sense. I'm new to Carla.

I'm trying to use the simulator in my undergraduate thesis, to validate a control law that I've developed. The problem is that the outputs of the control law are the wheels' angular speeds, since it was originally developed for robots with two wheels.

Reading the documentation for Carla, I found out that the parameters for vehiclecontrol are throttle, break, steer, .... Is there a way in which I can directly input the the wheel speed? I'm aware that this is not entirely realistic for a real application, but I would like to validate the control law that I developed.

Thanks in advance for the help!

jackbart94, 2020-02-13:
Hi,

I think that [this](https://carla.readthedocs.io/en/latest/how_to_control_vehicle_physics/) might help you. Make sure to also check the carla.VehiclePhysicsControl and carla.WheelPhysicsControl. Let me know if this helps you!

LuccaFaro, 2020-02-15:
Hi jackbart94,

Thanks for the answer. I read the link that you posted and the pages for both controls you mentioned. I'm not sure they let me control the angular speed for each wheel.

> Hi,
> 
> I think that [this](https://carla.readthedocs.io/en/latest/how_to_control_vehicle_physics/) might help you. Make sure to also check the carla.VehiclePhysicsControl and carla.WheelPhysicsControl. Let me know if this helps you!

surendrasah, 2020-03-18:
Hi @LuccaFaro ,
Even I am trying to do the same. I am not able to get the control of vehicle using speed. I donot need throttle. I am not sure how can we control the car by speed or wheels speed. Let me know if you have got a way to do this.
Hi @jackbart94 ,
I would really appreciate if you can advise/help on this.  carla.VehiclePhysicsControl and carla.WheelPhysicsControl are not helpful. I need to deal directly with speed/ wheel speed. 
I also have checked set_velocity() but I am not sure how do i implement this.

LuccaFaro, 2020-03-18:
Hi @surendrasah ,

I was having some trouble using Carla, so I'm trying another software, which seems to allow this more easily. Since I'm not very familiar with this kind of simulation, I'm having some trouble setting it up, so I cannot confirm it works yet.

I'm not sure I can share content about another software in this forum, but if you would like to know more about it, let me know!

surendrasah, 2020-03-18:
Hi @LuccaFaro 
Please share me the details of another software. I hope it might help me.

closed 2020-05-24 by stale[bot]

mindThomas, 2020-06-07:
Please also share the details of the software you are referring to with me at:
thomasj@tkjelectronics.dk
Thank you in advance.

CharlesXu1124, 2022-04-07:
I was wondering why are they so unclear about direct control via modifying wheel speed, is that something not inside cover and deep under Unreal engine?