carla-simulator/carla #2460: Stripes depth images [not smoothed]

animareversed, 2020-02-11:
Hi,
Another question: why depth images are striped? By any chance, is there a way to get smoothed depth images?
[image]

cassfalg, 2020-02-20:
How do you generate those? Did you read the docs on how to convert from carla's encoded format into meters or something similar? This looks like you just converted the rgb encoded image to grayscale, which is just wrong. Also read #2287 if you really need the depth data.

animareversed, 2020-02-20:
I did this:
 array = np.frombuffer(image.raw_data, dtype=np.dtype("uint8")) 
            array = np.reshape(array, (image.height, image.width, 4)) # RGBA format
            array = array[:, :, :3] #  Take only RGB
            array = array[:, :, ::-1] # BGR
            array = array.astype(np.float32) # 2ms
            gray_depth  = ((array[:,:,0] + array[:,:,1]*256.0 + array[:,:,2]*256.0*256.0)/((256.0*256.0*256.0) - 1)) # 2.5ms
            gray_depth = 1000 * gray_depth

This is what is written here, I think: https://carla.readthedocs.io/en/latest/cameras_and_sensors/#sensorcameradepth

No, I didn't convert the RGB image to grayscale... But it seems that all the depth images coming from simulator have these stripes (like VirtualKitti2 or Apollo synthetic), am I wrong?

cassfalg, 2020-02-24:
I have never tried to convert things directly from the image buffer, I always save them to disk. The saved images don't contain alpha for example. To me this banding looks like you chose the wrong colour channel order like BGR instead of RGB. You're also missing some multiplication characters in your code, not sure if that is a copy paste artefact.

What I can confirm is that the depth data in the saved images is correct after the fix for #2287 is applied. I have made pointclouds from depth and rgb images of the same frame, where a banding artefact like this would lead to very severe distortions that I would have noticed.

animareversed, 2020-02-25:
Well, I tried to save directly the depth from sensor and the saved images have alpha channel, as you can see in the image below:
[image]

This is exatcly the code that we used:
```
image_depth.convert(carla.ColorConverter.Depth)
array = np.frombuffer(image_depth.raw_data, dtype=np.dtype("uint8")) 
array = np.reshape(array, (image_depth.height, image_depth.width, 4)) # RGBA format
array = array[:, :, :3] #  Take only RGB
array = array[:, :, ::-1] # BGR
array = array.astype(np.float32) # 2ms
array  = ((array[:,:,0] + array[:,:,1]*256.0 + array[:,:,2]*256.0*256.0)/((256.0*256.0*256.0) - 1)) # 2.5ms
array = 1000 * array
```

I tried to follow this:
We calculate the center of the image as center = height/2 - 0.5
The focal length (distance to image plane in pixels) is: f = tan(fov/2) * heigh/2
The diagonal distance in pixels by pythagoras is: diagDist = sqrt((i-center)² + (j-center)² + f²)
The true depth of the pixel then is: depth * (diagDist / f)

but, which channel should I keep? I obtain these results:

1st channel:
[image]

2nd channel:
[image]

3rd channel:
[image]

cassfalg, 2020-02-25:
Not sure what this does:
```image_depth.convert(carla.ColorConverter.Depth)```
But it might do parts of the conversion in ways you do not want. I really don't know.

Regarding what is or is not part of the actual png image, I'd use imagemagick's identify command, maybe with the -verbose flag or so. Not sure if they use alpha, and what the "-1" flag to opencv does. That command however tells you number and depth of the channels.

minrui-hust, 2020-03-10:
have you solved this problem yet? I thing i encounter the same problem.

animareversed, 2020-03-10:
Sadly not.

minrui-hust, 2020-03-11:
It seems I found the reason,  it was weather asset not loaded correctly in may case issue #2572

closed 2020-05-17 by stale[bot]

lgalardini, 2021-04-11:
I think there are two separate things here (I'm quite new to Carla so please, if you think I'm wrong, please, correct me!). I hope it can help who came here without finding a solution.

One is related to the image with green/red with stripes above; if you use raw depth, the latter is stored using Unreal coordinate system. Please refer to the issue [#553](https://github.com/carla-simulator/carla/issues/553). 

Moreover, the depth in this case is not to be seen as a disparity map. If you want to obtain the real depth (in meters) you must use the following conversion, I noted this on my python code:
>         # Given a 2D point p2d = [u,v,1], your world point position P = [X,Y,Z] will be:
>         # P = ( inv(K) * p2d ) * depth

The second issue seems to be what @minrui-hust reported.

TzabarDolev, 2023-09-17:
There is an small error in the formula which is given in the 'readthedocs' of the depth sensor.
I wrote about it the CarlaSimBlog - https://carlasimblog.wordpress.com/2023/09/16/sensors/