carla-simulator/carla #2463: make PythonAPI does not work on Windows 10

xuyanwen2012, 2020-02-12:
Hello there,
My lab members and I have been stuck on this for many days already. 
We are trying to build Carla on Windows 10 machines, and the CarlaUE4 actually runs ok. ```make launch``` works perfectly, and Unreal is working, and we can play it. However, when trying to make the PythonAPI, ```make PythonAPI```, we have encountered many issues.

[image]

It is telling me Carla lib has been successfully installed but it is showing tons of linker errors. And I don't see any ```dist``` files been generated. 

And I therefore could not run the example python scripts.

[image]

- I am using the ```x64 Native Tools Command Prompt for VS 2017```
- I have ```Desktop Development with C++``` && ```Windows 8.1 SDK``` checked
- I am using ```Python 3.7.6``

Does anyone have encountered similar situations before? Please advise what should I do next.

Thank you very much.

ixfd64, 2020-02-14:
I'm also getting an error, although it's not the same.

`Traceback (most recent call last):
  File "setup.py", line 131, in <module>
    ext_modules=get_libcarla_extensions(),
  File "setup.py", line 91, in get_libcarla_extensions
    libs = [x for x in os.listdir('dependencies/lib') if any(d in x for d in required_libs)]
FileNotFoundError: [WinError 3] The system cannot find the path specified: 'dependencies/lib'`

Could it be related?

robogast, 2020-02-14:
I'm not sure if this exactly the issue, but it smells like it. See #1466.

The errors you show above smell specifically like errors with boost, although I'm not entirely sure what a fix for those would be.

Julian9875, 2020-03-01:
@xuyanwen2012 I have the exact same issue. Have you found the solution yet?

xuyanwen2012, 2020-03-02:
> @xuyanwen2012 I have the exact same issue. Have you found the solution yet?

Not yet, I am temporary getting around with this by switching to Linux (WSL). Although WSL have other issues that does not let me to launch the UE4editor, but at least the ``make PythonAPI`` can work. So I had my server running on Windows and the client running in WSL (kinda sucks, but at least it works).

j4l-Shvn, 2020-03-09:
Same here....seems like Windows was never built for driving simulator APIs :) ... possibly PACMan at the most.... Also, if we try the pypi version or the one that ships with the latest windows release i.e. carla 0.9.5 it throws the same "dependencies/lib" error..... made me believe I need to build it from scratch.. suffered miserably.... switched to linux instead...

carfloresp, 2020-04-04:
Hi @xuyanwen2012 , I faced the same problem and managed to get around it. After compiling Carla client and server, instead of "make PythonAPI", I did:

cd PythonAPI/carla
python setup.py install

This generated the /dist folder and the ".egg" file within. I am able to run the .py examples now.
Hope this helps in your case.

closed 2020-06-10 by stale[bot]

gn01121991, 2020-07-10:
Hi, I faced a similar issue recently, i solved the issue my opening the command prompt x64 Native Tools Command Prompt for VS 2017 with admin rights and followed steps mention in comment from @carfloresp ( path to carla\PythonAPI\carla and run python setup.py install)

kartikeyshaurya, 2020-10-09:
Thanks for the help i faced the similar issue now its working 💯

jzl20, 2020-11-17:
> elated

MarissaCampa, 2021-08-19:
> Hi @xuyanwen2012 , I faced the same problem and managed to get around it. After compiling Carla client and server, instead of "make PythonAPI", I did:
>
> cd PythonAPI/carla
> python setup.py install [...]

Thanks for the help @carfloresp , this method worked for me too!

arogyaGurkha, 2022-09-17:
> I'm also getting an error, although it's not the same.
>
> `Traceback (most recent call last): File "setup.py", line 131, in <module> ext_modules=get_libcarla_extensions(), File "setup.py", line 91, in [...]

Its been a few years but did you find way to solve this?

jzl20, 2022-09-27:
***@***.***

 
system can not find the dir   'dependencies/lib'
try to use the [full path] instead of [relative path]

抄送: " ***@***.******@***.***&gt;;

  

 
Traceback (most recent call last): File "setup.py", line 131, in <module&gt; ext_modules=get_libcarla_extensions(), File "setup.py", line 91, in get_libcarla_extensions libs = [x for x in os.listdir('dependencies/lib') if any(d in x for d in required_libs)] FileNotFoundError: [WinError 3] The system cannot find the path specified: 'dependencies/lib'

dhruv-g334, 2023-01-06:
> Hi @xuyanwen2012 , I faced the same problem and managed to get around it. After compiling Carla client and server, instead of "make PythonAPI", I did:
>
> cd PythonAPI/carla python setup.py install [...]

I am still facing the dependencies error. The dependencies folder is still not created.

Nighttell, 2023-01-10:
@dhruv-g334 Do you know how to solve it now?i met it today  :(

ixfd64, 2023-01-11:
> > I'm also getting an error, although it's not the same.
> > `Traceback (most recent call last): File "setup.py", line 131, in <module> ext_modules=get_libcarla_extensions(), File "setup.py", line 91, in get_libcarla_extensions libs = [x for x in os.listdir('dependencies/lib') if any(d in x for d in required_libs)] FileNotFoundError: [WinError 3] The system cannot find the path specified: 'dependencies/lib'`
> > Could it be related?
> 
> @ixfd64 Its been a few years but did you find way to solve this?

I don't remember the exact details as it's been a while. However, I think the issue went away for me after I upgraded to the latest release several months later.

Zainab-Saka, 2023-02-25:
hi @ixfd64 I have this same issue. How did you fix it Please?