carla-simulator/carla #2464: How to get Sensor ID for the sensor which generated some data?

shamitlal, 2020-02-12:
Hi,
How can I know which sensor generated some data? I am going through this example code.
https://github.com/cassfalg/carla/blob/master/PythonAPI/examples/synchronous_mode.py

My use-case requires me to attach multiple RGB and depth cameras to multiple vehicles. Therefore, 
I want to know the sensor ID for each generated datapoint. How can I achieve the same?

Thanks

rmohammedn, 2020-02-12:
Sensor ID doesn't change with time therefore, I think sensor data doesn't have sensor ID attribute. You can assign actor ID or some name to data when you are listening the sensors.

shamitlal, 2020-02-12:
So if I have multiple sensors all publishing data, how can I determine which sensor created a particular datapoint?

closed 2020-04-19 by stale[bot]