carla-simulator/carla #2470: WARNING: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API

EgorFollia, 2020-02-12:
How could I change a version of server/PythonApi ?
I got this after I running python script:
```
WARNING: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API 
WARNING: Client API version     = 0.9.7-30-g5a904579-dirty 
WARNING: Simulator API version  = 0.9.7-68-g3f6021df 

```

Georgini82, 2020-02-17:
Hello.
Did you updated carla?
Something happened to me because I updated carla but I didn't do "make PythonAPI" afterwords.
So my python API didn't match the version of my Simulator API.

I hope this helps you.
Regards

PS: Since both versions are 0.9.7-x it should be ok in terms of compatibility of APIs

EgorFollia, 2020-03-15:
@Georgini82 Hi, sorry for so late respond
Yes, I actually have reinstalled UE and Carla.
Should I use PythonApi from source folder or from Dist folder?

egreglee-bit, 2020-04-12:
This may be helpful: [ Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API](https://github.com/carla-simulator/carla/issues/2341).  Your Python client may be using an egg from another installation that it found via sys.path.

closed 2020-06-18 by stale[bot]

TrinhTuanHung2021, 2021-08-13:
> Hello.
> Did you updated carla?
> Something happened to me because I updated carla but I didn't do "make PythonAPI" afterwords.
> So my python API didn't match the version of my Simulator API.
>
> I hope [...]

Could you tell me about "make PythonAPI"? But when I run in the x64 Native Tools Command Prompt in the root directory folder, it has an error like this
[image]

tayeep, 2021-12-10:
> How could I change a version of server/PythonApi ? I got this after I running python script:
>
> ```
> WARNING: Version mismatch detected: You are trying to connect to a simulator that might be [...]

Simply delete the two Carla folders under C:\Users\<username>\AppData\Local\Programs\Python\Python37\Lib\site-packages\

The re-run the python script.

jorge2najo, 2023-10-05:
I encountered a similar issue when I installed Carla from source in one virtual environment (env1) and then set up Jupyter Notebook in another virtual environment (env2). In the second environment, I installed the 'carla' module using pip. Consequently, I kept receiving a warning whenever I attempted to import 'carla' in my Jupyter Notebook.

To resolve this, I first deleted 'env2' entirely. Then, I created a new virtual environment, 'env2_new,' where I installed Jupyter Notebook without adding the 'carla' module. Next, I copied the 'PythonAPI' folder and placed it in the directory where I intended to work with Jupyter Notebook.

To ensure that I could import the 'carla' module into my Jupyter Notebook projects, I added the following code snippet at the beginning of my Python file. You can find this snippet in the folder example of PythonAPI:

**import glob
import os
import sys
import time

try:
    sys.path.append(glob.glob('../<name_of_folder_outside_PythonAPI>/PythonAPI/carla/dist/carla-*%d.%d-%s.egg' % (
        sys.version_info.major,
        sys.version_info.minor,
        'win-amd64' if os.name == 'nt' else 'linux-x86_64'))[0])
except IndexError:
    pass

import carla**

The only adjustment you need to make is to ensure that the path to the 'dist' folder is correctly specified.