carla-simulator/carla #2472: trying to access an expired episode; a new episode was started in the simulation but an object tried accessing the old one

shamitlal, 2020-02-13:
Hi,
I am getting the following error when trying to randomly load a map in a loop:
"trying to access an expired episode; a new episode was started in the simulation but an object tried accessing the old one"

My carla version is 0.9.7.3.
Following is my code snippet:

`client = carla.Client(self.host, self.port)
  client.set_timeout(20.0)
  self.available_maps = client.get_available_maps()
  print("Available maps are: ", self.available_maps)
  logging.info('listening to server %s:%s', self.host, self.port)

        for episode in range(self.number_of_episodes):
            print("Starting episode number %d" %episode)
       
            cur_map = random.choice(self.available_maps)
            print("About to load the map %s" %cur_map)

            self.world = client.load_world(cur_map)
      
            # Get all the blueprints
            blueprints = self.world.get_blueprint_library().filter(self.filterv)
            
            print("printing blueprints")`

After loading the world in the second iteration, my code fails with the given error. Could you help me figure out what I might be doing wrong?

jackbart94, 2020-02-13:
Hi,

Technically, there's nothing you're doing wrong in this snippet. However, if there is anything going on in the first iteration that doesn't get destroyed properly, your actor (sensor, vehicle, whatever) is still trying to access the map you've already changed, and an exception is thrown upon changing an episode while having actors alive.

A solution would be to catch the exception and act accordingly, even ignore it, since we cleanup the episode upon changing map, or making sure every actor is destroyed before changing to a new map. 

Let me know if this solves your error!

shamitlal, 2020-02-13:
- But shouldn't load_world automatically destroy all the actors for the old world without me explicitly having to do so? Also, I tried maintaining list of all the actors I spawn (vehicles and sensors) and destroying them at the end but that didn't help as well. I also tried destroying all actors with is_alive flag set to true.

- It didn't give me this exception when I used carla simulator version 0.9.6!

cassfalg, 2020-02-20:
I think this is an error thrown by the server if a client still has a reference to an actor from an expired episode? So the error message is from the server's console ouput? Then you should look for expired references in your client script(s).

jackbart94, 2020-02-20:
Hi,

Unfortunately it will always throw this error because Traffic Manager still tries to access the actors. This has been fixed and is under review among many other things to be included in 0.9.8, or eventually a patch of 0.9.7, in the upcoming days!

closed 2020-04-30 by stale[bot]