carla-simulator/carla #2473: How to manually tune Autopilot parameters like speed, throttle etc.., My vehicle needs some harsh driving for my experimentation. Thanks in advance.

rabaig, 2020-02-13:


jackbart94, 2020-02-13:
Hi,

Please, provide more details regarding the version of CARLA you're using, since autopilot has been drastically changed in 0.9.7

rabaig, 2020-02-13:
Hi,

I am using carla_0.9.6 for my experiment

jackbart94, 2020-02-13:
Hi,

I suggest switching to 0.9.7 and using the Traffic Manager, which allows you to tune parameters for NPC vehicles, such as target speed among others.

rabaig, 2020-02-13:
My requirements :
1. setting a source and destination in a map.
2. manually tuning autopilot parameters to run my car in harsh drive mode .
3. Introducing more actors/ vehicles in my map with default autopilot mode along with my tuned vehicle.
4. Do i able to fetch data from all the vehicles in my map?

Can i able to do it with the 0.9.7 version?

please suggest

jackbart94, 2020-02-13:
Sure you can, there's plenty of examples that already do point 1,2 and 4 in the agents folder, while point 3 can be done through spawn_npc.py

rabaig, 2020-02-15:
hi,

I want to render the carla simulator on html page etc.. Is it possible

Thanks

rabaig, 2020-02-17:
To enable no rendering mode . I need to perform operations suggested in documentation
[CARLA/Server]
DisableRendering=true .   In windows carlas version 0.9.5 its seems CarlaSettings.ini file is missing.

could you please tell me where can i find  CarlaSettings.ini or in windows version it is supportable or not? 

Thanks

stevelaclasse, 2020-02-25:
i think the use of CarlaSettings.ini was with older versions of carla

rabaig, 2020-02-25:
hi,

I want to render 2D simulation of  the carla simulator on html page/possible UI  etc.. Is it possible

Thanks

closed 2020-05-02 by stale[bot]