carla-simulator/carla #2477: How to get list of actors in rbg image captured?

AbhijeetPendyala, 2020-02-14:
Hello guys, 

I am working on getting 2D bounding boxes from RGB image in addition to other actor( say vehicle) parameters like vehicle dimensions, centre of projection in BEV etc.

Is there a way to obtain the list of agents in the RGB image captured , contrary to full list of actors in the world  world.get_actors() ?

Thanks!

stevelaclasse, 2020-02-25:
i dont think you can obtain that so easily, you will have to do some calculations to remove actors that are not in the scene. you cant for example remove all the actors that are behind the scene and all those that are far from the scene and the that coordinates are not in the scene.

Harsharma2308, 2020-02-27:
Did you find anything useful on this?
Would be interesting to see if how this is figured out.
One small step could be to the 3D locations of the actors and see where are they located w.r.t to the camera. You could then remove most of the actors who are too far away in the left,right and behind from the car. You can just manually do it, or use the camera's FOV and distance information.

stevelaclasse, 2020-02-27:
you could check this issue 
[https://github.com/carla-simulator/carla/issues/2325](url)

Harsharma2308, 2020-03-06:
This doesn't answer this question

stevelaclasse, 2020-03-09:
this function return all the actor in the image : https://github.com/carla-simulator/carla/issues/2325#issuecomment-582934474

have you tested it ?? sure there are values that you could edit, accordingly to what you want.

Harsharma2308, 2020-03-20:
Sorry if i wasnt clear.. if i understood correctly that script checks if a bounding box is too small, and discards it if it is.
But doesn't take care of the occlusions nearby. 
(When the camera is facing towards a straight road you would not want to reject far away boxes but would rather want to remove the boxes behind a building)

stevelaclasse, 2020-03-24:
Hi, please could you comment it on the Issue on Github, it will be easier
for me to answer. Thanks

stevelaclasse, 2020-03-24:
> 
> 
> Sorry if i wasnt clear.. if i understood correctly that script checks if a bounding box is too small, and discards it if it is.
> But doesn't take care of the occlusions nearby.
> (When the camera is facing towards a straight road you would not want to reject far away boxes but would rather want to remove the nearby boxes which are probably from vehicles behind a building)

yes, it rejects small boxes but don't care of occlusions.  i don't have for now a way of detecting if a vehicle is occluded.

closed 2020-05-30 by stale[bot]

jonbakerfish, 2021-11-14:
@Pendu Did you solve this problem?

stevelaclasse, 2021-11-15:
i have train an Object detection model on the images captured by the RGB Camera (in three classes: "cars","pedestrians" and "occulted"), then i have use this model to test each box on the image i got from the RGB Camera.  
Annother way could be to use the **Depth Camera** to get the  distance of objets in the images.

the question was not for me, but i hope a help a bit.

jonbakerfish, 2021-11-16:
@stevelaclasse Thanks. 

FYI: just find a good repo that explains how to do this https://mukhlasadib.github.io/CARLA-2DBBox/

luke3322, 2024-04-03:
> i have train an Object detection model on the images captured by the RGB Camera (in three classes: "cars","pedestrians" and "occulted"), then i have use this model to test each box on the image i [...]

Hi, I am about to realize a very similiar project. Did you manage do get the information of the vehicles in your pictures?