carla-simulator/carla #2490: CARLA 0.9.7.3 with Vulkan totally fails

najibghadri, 2020-02-19:
- Linux, Ubuntu 18.04, x86_64
- NVIDIA MX150
- NVIDIA driver version 
- some important packages that I already have (both in amd64 and i386):
```
libnvidia-gl-440                     440.59-0ubuntu0.18.04.1
libnvidia-compute-440                440.59-0ubuntu0.18.04.1
libvulkan1                           1.1.126.0-2~gpu18.04.1
vulkan-tools                               1.1.126.0+dfsg1-1~gpu18.04.1
vulkan-utils                               1.1.126.0+dfsg1-1~gpu18.04.1
mesa-vulkan-drivers                  20.1~git2002190730.72f7d3~oibaf~b
```
and everything I tought would be useful..

I can run it in -opengl. But I want to run in with vulkan, however this is the error:
```
>> ./CarlaUE4.sh 
4.22.1-0+++UE4+Release-4.22 517 0
Disabling core dumps.
X Error of failed request:  BadDrawable (invalid Pixmap or Window parameter)
  Major opcode of failed request:  149 ()
  Minor opcode of failed request:  4
  Resource id in failed request:  0x3a00041
  Serial number of failed request:  363
  Current serial number in output stream:  369
Signal 11 caught.
Segmentation fault (core dumped)
```
I have reinstalled many packeges, rebooted my pc, etc.
If you know for  that another prev version is gonna work I can try, but I won't download each 3 gig carla version if the solution is not lying there.

Thanks in advance!

germanros1987, 2020-02-20:
Uhm, I am not sure if that could have something to do with the specific GPU you are using (MX150). Could you try with a previous version, for instance, 0.9.6?

najibghadri, 2020-02-20:
Yeah I too assume it doesn't have much to do with the gpu, rather the driver and Carla's communication with the driver through vulkan. Maybe it's a vulkan problem, I have no idea, I might try another version, but as for now this version fails. I will update, I feel like I can live with OpenGL..

najibghadri, 2020-02-20:
Actually I tried with the open source nouveau drivers and it works with that! But obviously it was very slow since those drivers aren't optimized for the gpu (is what I imagine). The previous nvidia drivers didn't even work with the driver. Hence it is surely a driver compatibility problem

closed 2020-05-25 by stale[bot]