carla-simulator/carla #2495: Oscillating vehicles in 0.9.7

ndeshp2s, 2020-02-20:
I have recently shifted to carla 0.9.7 version and I noticed the vehicles oscillate a lot while taking turns or even while passing straight through junctions. I tried tuning the PID (lateral control) parameters in local_planner.py file. But this makes no impact on this oscillating behavior.

Is there a way to resolve this? Am I missing something?

Thank you.

jackbart94, 2020-02-20:
Hi,

Tuning the parameters in local_planner.py only affects the agents, but if you're using the Traffic Manager, activated by set_autopilot(True), you need to change those PID parameters. However, for FPS under 20, this oscillating behavior happens anyway.

You can try to tune them in TrafficManager.cpp, but in a few days this will be fixed by introducing synchronous mode with Traffic Manager!

SunFishlike, 2020-02-27:
@jackbart94 Hi,

I have been using carla since 0.9.6, but I'm a little bit confused. Can you give a clue about how to synchronize? 

Using the provided synchronous mode example, I managed to get all things well previously in 0.9.6 WITH NPC vehicles, . But in 0.9.7.4, NPC vehicles are abnormal with previous synchronous example. I think it may be because of synchronous problem of traffic manager, or there should be some method to activate the synchronous mode for it, but I didn't see there's any method inside carla.libcarla.TrafficManager .

jackbart94, 2020-02-27:
@SunFishlike Hi!

There was a big change from 0.9.6 to 0.9.7 when it comes to handling vehicles. In 0.9.8 (coming out in two weeks) we have fixed the NPC abnormal behavior and reintroduced a working synchronous mode compliant with Traffic Manager. 

But right now, in 0.9.7, the Traffic Manager is asynchronous and thus a big drop in FPS will cause vehicles to oscillate, because they are skipping frames.

We are merging these changes in master soon, you will be able to try it in the upcoming days!

martonur, 2020-05-04:
Hi guys!
The oscillating behavior is still present for me even if I set CARLA to synchronous mode and fixed timestep. (using the new, 0.9.9 version and Traffic Manager for autopilot)

germanros1987, 2020-06-26:
@jackbart94 this issue needs attention. Are you able to reproduce this behavior in TM 2.0?
@martonur could you please provide a minimum example?

jackbart94, 2020-06-26:
Hi,

I haven't been able to see vehicles oscillating in TM 2.0. I need more information about this in order to reproduce it.

closed 2020-09-05 by stale[bot]

hangqiu, 2020-10-20:
Has this issue been solved? I can confirm that the issue persists in carla 0.9.9 + scenario runner 0.9.9 in town03

martonur, 2020-10-22:
I will test it with 0.9.10 and if the problem is still present, I'll provide a minimal example for easy reproduction.

hangqiu, 2020-11-05:
I can confirm this still happens in carla 0.9.10. The observations are that lane changing cars will spin, straight going cars will weave in the lane. I can try but I believe It's tough for me to generate a minimal reproducible example from my current stack. But the observation was made when there are a lot of traffics and sensors on each of them and lots of computations on client side that makes the server run much less than 1fps.