carla-simulator/carla #2497: PropFactory not working in Carla 0.9.7

Georgini82, 2020-02-20:
Hello.

I'm trying to add a new prop to Carla.
But when I update the PropFactory with a new static mesh it doesn't show the blueprint in the world available list of BP.

Can anyone confirm if this is an issue or if I'm missing something please?

OS: Xubuntu 18.04
UE4: 4.22
Carla: 0.9.7

a-jahani, 2020-02-20:
Yeah, the Prop factory blueprint won't work. You need to add a JSON file for the prop.

Georgini82, 2020-02-26:
Thanks for the reply a-jahani.

Going to try it.

Regards

closed 2020-02-26 by Georgini82

omrizen, 2020-05-13:
on carla 0.9.8 the tag does not work properly for me when I get an image from semantic segmentation image
(the color of the object is black)
what are the options for tagging that I can put? 
Thanks