carla-simulator/carla #2499: make launch C++17 error windows

savinovava, 2020-02-21:
[image]
```
 LocalizationStage.cpp
  class template optional is only available with C++17 or later.
and after a lot of errors,
```
i'm using VS 2017

[image]

I have downloaded C++/WinRT but it doesn't help

jackbart94, 2020-02-24:
Hi,

Sorry for the issue, one of the variables is declared as `uint` instead of `uint64_t`. It's on the line of the error, changing that should solve your problem.

JimWhiteheadMR, 2020-02-25:
Just a quick note to say that I also ran into this same issue (uint instead of specific width integer) with LocalizationStage.cpp, and in my case changing all instances of uint to uint32_t in LocalizationStage.cpp fixed the problem.

I also ran into a similar issue with ProfilesParser.cpp. Here I changed all variable declarations of uint to uint32_t, and this fixed the issue. Note that in ProfilesParser.cpp, I did not change lines of the form:

node_road.attribute("id").as_uint();

These were left as-is, since there is no as_uint32_t() equivalent.

I am compiling on Windows 10, using Visual Studio 2017.

savinovava, 2020-03-28:
I decide to do linux build, so can`t check the recomendations now

closed 2020-03-28 by savinovava