carla-simulator/carla #2501: I encountered an error while building this version.

ClintCPvis, 2020-02-22:
clint@clint-Alienware-m15:~/carla$ make PythonAPI && cd PythonAPI/examples && python tm_spawn_npc.py
Setup.sh: llvm-7.0 already installed.
Setup.sh: boost-1.69.0-c7 already installed.
Setup.sh: rpclib-v2.2.1_c2-c7 already installed.
Setup.sh: gtest-1.8.1-c7 already installed.
Setup.sh: recast-cdce4e-c7 already installed.
Setup.sh: CARLA version 0.9.7-80-g731451d4.
Setup.sh: Generating CMake configuration files.
Setup.sh: Success!
BuildLibCarla.sh: Building LibCarla "Client.Release" configuration.
+ cmake -G 'Eclipse CDT4 - Ninja' -DCMAKE_BUILD_TYPE=Client -DLIBCARLA_BUILD_DEBUG=OFF -DLIBCARLA_BUILD_RELEASE=ON -DCMAKE_TOOLCHAIN_FILE=/home/clint/carla/Build/LibStdCppToolChain.cmake -DCMAKE_INSTALL_PREFIX=/home/clint/carla/PythonAPI/carla/dependencies -DCMAKE_EXPORT_COMPILE_COMMANDS=1 /home/clint/carla
-- The C compiler identification is Clang 7.1.0
-- The CXX compiler identification is Clang 7.1.0
-- Could not determine Eclipse version, assuming at least 3.6 (Helios). Adjust CMAKE_ECLIPSE_VERSION if this is wrong.
-- Check for working C compiler: /usr/bin/clang-7
-- Check for working C compiler: /usr/bin/clang-7 -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/clang++-7
-- Check for working CXX compiler: /usr/bin/clang++-7 -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Build debug:   OFF
-- Build release: ON
-- Build test:    ON
-- Configuring done
CMake Warning in CMakeLists.txt:
  The build directory is a subdirectory of the source directory.

  This is not supported well by Eclipse.  It is strongly recommended to use a
  build directory which is a sibling of the source directory.

-- Generating done
-- Build files have been written to: /home/clint/carla/Build/libcarla-client-build.release
+ ninja
ninja: error: '../boost-1.69.0-c7-install/lib/libboost_filesystem.a', needed by 'LibCarla/cmake/test/libcarla_test_client_release', missing and no known rule to make it
Util/BuildTools/Linux.mk:117: recipe for target 'LibCarla.client.release' failed
make: *** [LibCarla.client.release] Error 1
clint@clint-Alienware-m15:~/carla$

jackbart94, 2020-02-24:
Hi,

Did you do 'make LibCarla' first?

ClintCPvis, 2020-02-24:
I did not, I suppose that could be the problem. I only followed these steps: 
https://carla.readthedocs.io/en/latest/how_to_build_on_linux/
[I](url) will start over and try that now.

ClintCPvis, 2020-02-24:
> Hi,
> 
> Did you do `make LibCarla` first?

~/carla$ make LibCarla
Setup.sh: llvm-7.0 already installed.
Setup.sh: boost-1.69.0-c7 already installed.
Setup.sh: rpclib-v2.2.1_c2-c7 already installed.
Setup.sh: gtest-1.8.1-c7 already installed.
Setup.sh: recast-cdce4e-c7 already installed.
Setup.sh: CARLA version 0.9.7-81-g06b5a3d9.
Setup.sh: Generating CMake configuration files.
Setup.sh: Success!
BuildLibCarla.sh: Building LibCarla "Server.Release" configuration.

...

    typename std::enable_if<boost::asio::is_const_buffer_sequence<T>::value>::type
                            ~~~~~~~~~~~~~^
../../LibCarla/cmake/../source/carla/Buffer.h:308:67: error: 'T' does not refer to a value
    typename std::enable_if<boost::asio::is_const_buffer_sequence<T>::value>::type
                                                                  ^
../../LibCarla/cmake/../source/carla/Buffer.h:307:24: note: declared here
    template <typename T>
                       ^
../../LibCarla/cmake/../source/carla/Buffer.h:308:71: error: member 'value' declared as a template
    typename std::enable_if<boost::asio::is_const_buffer_sequence<T>::value>::type
                                                                      ^
../../LibCarla/cmake/../source/carla/Buffer.h:308:76: error: expected ';' at end of declaration list
    typename std::enable_if<boost::asio::is_const_buffer_sequence<T>::value>::type
                                                                           ^
                                                                           ;
../../LibCarla/cmake/../source/carla/Buffer.h:111:9: error: use of undeclared identifier 'ReuseThisBuffer'
        ReuseThisBuffer();
        ^
../../LibCarla/cmake/../source/carla/Buffer.h:297:7: error: no matching member function for call to 'copy_from'
      copy_from(0u, data, size);
      ^~~~~~~~~
../../LibCarla/cmake/../source/carla/Buffer.h:290:10: note: candidate function template not viable: requires single argument 'source', but 3 arguments were provided
    void copy_from(const T &source) {
         ^
../../LibCarla/cmake/../source/carla/Buffer.h:302:10: note: candidate function not viable: requires 2 arguments, but 3 were provided
    void copy_from(size_type offset, const Buffer &rhs) {
         ^
../../LibCarla/cmake/../source/carla/Buffer.h:296:10: note: candidate function not viable: requires 2 arguments, but 3 were provided
    void copy_from(const value_type *data, size_type size) {
         ^
../../LibCarla/cmake/../source/carla/Buffer.h:303:7: error: no matching member function for call to 'copy_from'
      copy_from(offset, rhs.buffer());
      ^~~~~~~~~
../../LibCarla/cmake/../source/carla/Buffer.h:302:10: note: candidate function not viable: no known conversion from 'boost::asio::const_buffer' to 'const carla::Buffer' for 2nd argument
    void copy_from(size_type offset, const Buffer &rhs) {
         ^
../../LibCarla/cmake/../source/carla/Buffer.h:296:10: note: candidate function not viable: no known conversion from 'carla::Buffer::size_type' (aka 'unsigned int') to 'const carla::Buffer::value_type *' (aka 'const unsigned char *') for 1st argument
    void copy_from(const value_type *data, size_type size) {
         ^
../../LibCarla/cmake/../source/carla/Buffer.h:290:10: note: candidate function template not viable: requires single argument 'source', but 2 arguments were provided
    void copy_from(const T &source) {
         ^
7 errors generated.
ninja: build stopped: subcommand failed.
Util/BuildTools/Linux.mk:111: recipe for target 'LibCarla.server.release' failed
make: *** [LibCarla.server.release] Error 1

this is the error i got. im not sure what the issue could be.

closed 2020-05-02 by stale[bot]

vishal-246, 2020-05-26:
Hi,

while building PythonAPI and LibCarla i got the similar kind of issue, below is log 

> vishal@D-11301:/carla/carla$ make PythonAPI
Setup.sh: llvm-8.0 already installed.
Setup.sh: boost-1.72.0-c8 already installed.
Setup.sh: rpclib-v2.2.1_c2-c8 already installed.
Setup.sh: gtest-1.8.1-c8 already installed.
Setup.sh: recast-cdce4e-c8 already installed.
Setup.sh: CARLA version 0.9.9-58-g4dc4cb81-dirty.
Setup.sh: Generating CMake configuration files.
Setup.sh: Success!
BuildLibCarla.sh: Building LibCarla "Client.Release" configuration.
[1/111] Building CXX object LibCarla/cmake/client/CMakeFiles/carla_client.dir/__/__/source/carla/client/Actor.cpp.o
FAILED: LibCarla/cmake/client/CMakeFiles/carla_client.dir/__/__/source/carla/client/Actor.cpp.o 
/usr/bin/clang++-8  -DBOOST_ERROR_CODE_HEADER_ONLY -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true -DLIBCARLA_TEST_CONTENT_FOLDER=\"/carla/carla/Build/test-content\" -I../../LibCarla/cmake/../source -I../../LibCarla/cmake/../source/third-party -isystem ../boost-1.72.0-c8-install/include -isystem ../rpclib-v2.2.1_c2-c8-libstdcxx-install/include -isystem ../recast-cdce4e-c8-install/include -std=c++14 -pthread -fPIC -Werror -Wall -Wextra -Wpedantic -Wdeprecated -Wshadow -Wuninitialized -Wunreachable-code -Wpessimizing-move -Wold-style-cast -Wnull-dereference -Wduplicate-enum -Wnon-virtual-dtor -Wheader-hygiene -Wconversion -Wfloat-overflow-conversion -std=c++14 -pthread -fPIC -Werror -Wall -Wextra -Wpedantic -Wdeprecated -Wshadow -Wuninitialized -Wunreachable-code -Wpessimizing-move -Wold-style-cast -Wnull-dereference -Wduplicate-enum -Wnon-virtual-dtor -Wheader-hygiene -Wconversion -Wfloat-overflow-conversion    -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -Wdate-time -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -std=c++14 -Wno-missing-braces -DBOOST_ERROR_CODE_HEADER_ONLY -MD -MT LibCarla/cmake/client/CMakeFiles/carla_client.dir/__/__/source/carla/client/Actor.cpp.o -MF LibCarla/cmake/client/CMakeFiles/carla_client.dir/__/__/source/carla/client/Actor.cpp.o.d -o LibCarla/cmake/client/CMakeFiles/carla_client.dir/__/__/source/carla/client/Actor.cpp.o -c ../../LibCarla/source/carla/client/Actor.cpp
In file included from ../../LibCarla/source/carla/client/Actor.cpp:7:
In file included from ../../LibCarla/cmake/../source/carla/client/Actor.h:11:
In file included from ../../LibCarla/cmake/../source/carla/client/detail/ActorState.h:10:
In file included from ../../LibCarla/cmake/../source/carla/client/World.h:13:
In file included from ../../LibCarla/cmake/../source/carla/client/LightManager.h:17:
In file included from ../../LibCarla/cmake/../source/carla/client/detail/Episode.h:17:
In file included from ../../LibCarla/cmake/../source/carla/client/detail/WalkerNavigation.h:10:
../../LibCarla/cmake/../source/carla/nav/Navigation.h:16:10: fatal error: 'recast/Recast.h' file not found
#include <recast/Recast.h>
         ^~~~~~~~~~~~~~~~~
1 error generated.
[2/111] Building CXX object LibCarla/cmake/client/CMakeFiles/carla_client.dir/__/__/source/carla/client/DebugHelper.cpp.o
FAILED: LibCarla/cmake/client/CMakeFiles/carla_client.dir/__/__/source/carla/client/DebugHelper.cpp.o 
/usr/bin/clang++-8  -DBOOST_ERROR_CODE_HEADER_ONLY -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true -DLIBCARLA_TEST_CONTENT_FOLDER=\"/carla/carla/Build/test-content\" -I../../LibCarla/cmake/../source -I../../LibCarla/cmake/../source/third-party -isystem ../boost-1.72.0-c8-install/include -isystem ../rpclib-v2.2.1_c2-c8-libstdcxx-install/include -isystem ../recast-cdce4e-c8-install/include -std=c++14 -pthread -fPIC -Werror -Wall -Wextra -Wpedantic -Wdeprecated -Wshadow -Wuninitialized -Wunreachable-code -Wpessimizing-move -Wold-style-cast -Wnull-dereference -Wduplicate-enum -Wnon-virtual-dtor -Wheader-hygiene -Wconversion -Wfloat-overflow-conversion -std=c++14 -pthread -fPIC -Werror -Wall -Wextra -Wpedantic -Wdeprecated -Wshadow -Wuninitialized -Wunreachable-code -Wpessimizing-move -Wold-style-cast -Wnull-dereference -Wduplicate-enum -Wnon-virtual-dtor -Wheader-hygiene -Wconversion -Wfloat-overflow-conversion    -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -Wdate-time -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -std=c++14 -Wno-missing-braces -DBOOST_ERROR_CODE_HEADER_ONLY -MD -MT LibCarla/cmake/client/CMakeFiles/carla_client.dir/__/__/source/carla/client/DebugHelper.cpp.o -MF LibCarla/cmake/client/CMakeFiles/carla_client.dir/__/__/source/carla/client/DebugHelper.cpp.o.d -o LibCarla/cmake/client/CMakeFiles/carla_client.dir/__/__/source/carla/client/DebugHelper.cpp.o -c ../../LibCarla/source/carla/client/DebugHelper.cpp
In file included from ../../LibCarla/source/carla/client/DebugHelper.cpp:9:
In file included from ../../LibCarla/cmake/../source/carla/client/detail/Simulator.h:12:
In file included from ../../LibCarla/cmake/../source/carla/client/Actor.h:11:
In file included from ../../LibCarla/cmake/../source/carla/client/detail/ActorState.h:10:
In file included from ../../LibCarla/cmake/../source/carla/client/World.h:13:
In file included from ../../LibCarla/cmake/../source/carla/client/LightManager.h:17:
In file included from ../../LibCarla/cmake/../source/carla/client/detail/Episode.h:17:
In file included from ../../LibCarla/cmake/../source/carla/client/detail/WalkerNavigation.h:10:
[... cut ...]
../../LibCarla/cmake/../source/carla/nav/Navigation.h:16:10: fatal error: 'recast/Recast.h' file not found
#include <recast/Recast.h>
         ^~~~~~~~~~~~~~~~~
1 error generated.
[6/111] Building CXX object LibCarla/cmake/client/CMakeFiles/carla_client.dir/__/__/source/carla/client/Junction.cpp.o
ninja: build stopped: subcommand failed.
Util/BuildTools/Linux.mk:126: recipe for target 'LibCarla.client.release' failed
make: *** [LibCarla.client.release] Error 1
vishal@D-11301:/carla/carla$ 

please help me to solve this .!

Thank you.

schaefma3, 2020-06-12:
I have exactly the same issue. 

Thanks
Max

Abhishek-Chachadi, 2020-07-07:
make launch
Setup.sh: llvm-8.0 already installed.
Setup.sh: boost-1.72.0-c8 already installed.
Setup.sh: rpclib-v2.2.1_c2-c8 already installed.
Setup.sh: gtest-1.8.1-c8 already installed.
Setup.sh: recast-cdce4e-c8 already installed.
Setup.sh: CARLA version 0.9.9.4-25-g6ea32e4-dirty.
Setup.sh: Generating CMake configuration files.
Setup.sh: Success!
BuildLibCarla.sh: Building LibCarla "Server.Release" configuration.
[6/51] Building CXX object LibCarla/cm...er.dir/__/__/source/carla/Buffer.cpp.o
FAILED: /usr/bin/clang++-8 -DASIO_NO_EXCEPTIONS -DBOOST_ERROR_CODE_HEADER_ONLY -DBOOST_NO_EXCEPTIONS -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true -DLIBCARLA_NO_EXCEPTIONS -DLIBCARLA_TEST_CONTENT_FOLDER=\"/media/abhishek/202A186C1F59827F/UnrealEngine_4.24/Engine/Binaries/Linux/carla/Build/test-content\" -DPUGIXML_NO_EXCEPTIONS -I../../LibCarla/cmake/../source -I../../LibCarla/cmake/../source/third-party -isystem ../boost-1.72.0-c8-install/include -isystem ../rpclib-v2.2.1_c2-c8-libcxx-install/include -std=c++14 -pthread -fPIC -Werror -Wall -Wextra -Wpedantic -Wdeprecated -Wshadow -Wuninitialized -Wunreachable-code -Wpessimizing-move -Wold-style-cast -Wnull-dereference -Wduplicate-enum -Wnon-virtual-dtor -Wheader-hygiene -Wconversion -Wfloat-overflow-conversion -stdlib=libc++ -isystem /media/abhishek/202A186C1F59827F/UnrealEngine_4.24/Engine/Binaries/Linux/carla/Build/llvm-8.0-install/include/c++/v1 -fno-exceptions -fno-rtti -std=c++14 -pthread -fPIC -Werror -Wall -Wextra -Wpedantic -Wdeprecated -Wshadow -Wuninitialized -Wunreachable-code -Wpessimizing-move -Wold-style-cast -Wnull-dereference -Wduplicate-enum -Wnon-virtual-dtor -Wheader-hygiene -Wconversion -Wfloat-overflow-conversion -stdlib=libc++ -isystem /media/abhishek/202A186C1F59827F/UnrealEngine_4.24/Engine/Binaries/Linux/carla/Build/llvm-8.0-install/include/c++/v1 -fno-exceptions -fno-rtti -O3 -DNDEBUG -MD -MT LibCarla/cmake/server/CMakeFiles/carla_server.dir/__/__/source/carla/Buffer.cpp.o -MF LibCarla/cmake/server/CMakeFiles/carla_server.dir/__/__/source/carla/Buffer.cpp.o.d -o LibCarla/cmake/server/CMakeFiles/carla_server.dir/__/__/source/carla/Buffer.cpp.o -c ../../LibCarla/source/carla/Buffer.cpp
In file included from ../../LibCarla/source/carla/Buffer.cpp:1:
../../LibCarla/cmake/../source/carla/Buffer.h:320:42: error: no member named 'is_const_buffer_sequence' in namespace 'boost::asio'
    typename std::enable_if<boost::asio::is_const_buffer_sequence<T>::value>::type
                            ~~~~~~~~~~~~~^
../../LibCarla/cmake/../source/carla/Buffer.h:320:67: error: 'T' does not refer to a value
    typename std::enable_if<boost::asio::is_const_buffer_sequence<T>::value>::type
                                                                  ^
../../LibCarla/cmake/../source/carla/Buffer.h:319:24: note: declared here
    template <typename T>
                       ^
../../LibCarla/cmake/../source/carla/Buffer.h:320:71: error: member 'value' declared as a template
    typename std::enable_if<boost::asio::is_const_buffer_sequence<T>::value>::type
                                                                      ^
../../LibCarla/cmake/../source/carla/Buffer.h:320:76: error: expected ';' at end of declaration list
    typename std::enable_if<boost::asio::is_const_buffer_sequence<T>::value>::type
                                                                           ^
                                                                           ;
../../LibCarla/cmake/../source/carla/Buffer.h:111:9: error: use of undeclared identifier 'ReuseThisBuffer'
        ReuseThisBuffer();
        ^
../../LibCarla/cmake/../source/carla/Buffer.h:309:7: error: no matching member function for call to 'copy_from'
      copy_from(0u, data, size);
      ^~~~~~~~~
../../LibCarla/cmake/../source/carla/Buffer.h:302:10: note: candidate function template not viable: requires single argument 'source', but 3 arguments were provided
    void copy_from(const T &source) {
         ^
../../LibCarla/cmake/../source/carla/Buffer.h:314:10: note: candidate function not viable: requires 2 arguments, but 3 were provided
    void copy_from(size_type offset, const Buffer &rhs) {
         ^
../../LibCarla/cmake/../source/carla/Buffer.h:308:10: note: candidate function not viable: requires 2 arguments, but 3 were provided
    void copy_from(const value_type *data, size_type size) {
         ^
../../LibCarla/cmake/../source/carla/Buffer.h:315:7: error: no matching member function for call to 'copy_from'
      copy_from(offset, rhs.buffer());
      ^~~~~~~~~
../../LibCarla/cmake/../source/carla/Buffer.h:314:10: note: candidate function not viable: no known conversion from 'boost::asio::const_buffer' to 'const carla::Buffer' for 2nd argument
    void copy_from(size_type offset, const Buffer &rhs) {
         ^
../../LibCarla/cmake/../source/carla/Buffer.h:308:10: note: candidate function not viable: no known conversion from 'carla::Buffer::size_type' (aka 'unsigned int') to 'const carla::Buffer::value_type *' (aka 'const unsigned char *') for 1st argument
    void copy_from(const value_type *data, size_type size) {
         ^
../../LibCarla/cmake/../source/carla/Buffer.h:302:10: note: candidate function template not viable: requires single argument 'source', but 2 arguments were provided
    void copy_from(const T &source) {
         ^
../../LibCarla/source/carla/Buffer.cpp:7:16: error: out-of-line definition of 'ReuseThisBuffer' does not match any declaration in 'carla::Buffer'
[... cut ...]
    void copy_from(const value_type *data, size_type size) {
         ^
../../LibCarla/cmake/../source/carla/Buffer.h:302:10: note: candidate function template not viable: requires single argument 'source', but 2 arguments were provided
    void copy_from(const T &source) {
         ^
7 errors generated.
ninja: build stopped: subcommand failed.
Util/BuildTools/Linux.mk:120: recipe for target 'LibCarla.server.release' failed
make: *** [LibCarla.server.release] Error 1

I have a same issue here
Thanx

ArghyaPal, 2020-08-15:
Same problem. Any help

SamStaps, 2020-08-18:
Install python-dev. Delete the build directory. Repeat the make process.

godture, 2020-09-21:
> > Hi,
> > Did you do `make LibCarla` first?
> 
> ~/carla$ make LibCarla
> Setup.sh: llvm-7.0 already installed.
> Setup.sh: boost-1.69.0-c7 already installed.
> Setup.sh: rpclib-v2.2.1_c2-c7 already installed.
> Setup.sh: gtest-1.8.1-c7 already installed.
> Setup.sh: recast-cdce4e-c7 already installed.
> Setup.sh: CARLA version 0.9.7-81-g06b5a3d9.
> Setup.sh: Generating CMake configuration files.
> Setup.sh: Success!
> BuildLibCarla.sh: Building LibCarla "Server.Release" configuration.
> 
> ...
> 
> ```
> typename std::enable_if<boost::asio::is_const_buffer_sequence<T>::value>::type
>                         ~~~~~~~~~~~~~^
> ```
> 
> ../../LibCarla/cmake/../source/carla/Buffer.h:308:67: error: 'T' does not refer to a value
> typename std::enable_if<boost::asio::is_const_buffer_sequence::value>::type
> ^
> ../../LibCarla/cmake/../source/carla/Buffer.h:307:24: note: declared here
> template
> ^
> ../../LibCarla/cmake/../source/carla/Buffer.h:308:71: error: member 'value' declared as a template
> typename std::enable_if<boost::asio::is_const_buffer_sequence::value>::type
> ^
> ../../LibCarla/cmake/../source/carla/Buffer.h:308:76: error: expected ';' at end of declaration list
> typename std::enable_if<boost::asio::is_const_buffer_sequence::value>::type
> ^
> ;
> ../../LibCarla/cmake/../source/carla/Buffer.h:111:9: error: use of undeclared identifier 'ReuseThisBuffer'
> ReuseThisBuffer();
> ^
> ../../LibCarla/cmake/../source/carla/Buffer.h:297:7: error: no matching member function for call to 'copy_from'
> copy_from(0u, data, size);
> ^~~~~~~~~
> ../../LibCarla/cmake/../source/carla/Buffer.h:290:10: note: candidate function template not viable: requires single argument 'source', but 3 arguments were provided
> void copy_from(const T &source) {
> ^
> ../../LibCarla/cmake/../source/carla/Buffer.h:302:10: note: candidate function not viable: requires 2 arguments, but 3 were provided
> void copy_from(size_type offset, const Buffer &rhs) {
> ^
> ../../LibCarla/cmake/../source/carla/Buffer.h:296:10: note: candidate function not viable: requires 2 arguments, but 3 were provided
> void copy_from(const value_type *data, size_type size) {
> ^
> ../../LibCarla/cmake/../source/carla/Buffer.h:303:7: error: no matching member function for call to 'copy_from'
> copy_from(offset, rhs.buffer());
> ^~~~~~~~~
> ../../LibCarla/cmake/../source/carla/Buffer.h:302:10: note: candidate function not viable: no known conversion from 'boost::asio::const_buffer' to 'const carla::Buffer' for 2nd argument
> void copy_from(size_type offset, const Buffer &rhs) {
> ^
> ../../LibCarla/cmake/../source/carla/Buffer.h:296:10: note: candidate function not viable: no known conversion from 'carla::Buffer::size_type' (aka 'unsigned int') to 'const carla::Buffer::value_type *' (aka 'const unsigned char *') for 1st argument
> void copy_from(const value_type *data, size_type size) {
> ^
> ../../LibCarla/cmake/../source/carla/Buffer.h:290:10: note: candidate function template not viable: requires single argument 'source', but 2 arguments were provided
> void copy_from(const T &source) {
> ^
> 7 errors generated.
> ninja: build stopped: subcommand failed.
> Util/BuildTools/Linux.mk:111: recipe for target 'LibCarla.server.release' failed
> make: *** [LibCarla.server.release] Error 1
> 
> this is the error i got. im not sure what the issue could be.

I have exactly the same problem. Not solved yet.

godture, 2020-09-21:
> Install python-dev. Delete the build directory. Repeat the make process.

This does not work. I have python-dev already installed.