carla-simulator/carla #2503: error CARLA: gzip: stdin: not in gzip format tar

SBM1952, 2020-02-24:
Hi,
I've followed the steps above an i m getting these prompts:

After Update.sh

Content.tar.gz      100%[===================>]   9,25G  5,93MB/s    in 27m 19s 

2020-02-24 09:00:44 (5,78 MB/s) - ‘Content.tar.gz’ saved [9936321834/9936321834]
gzip: stdin: not in gzip format
tar: Child returned status 1
tar: Error is not recoverable: exiting now

after building the target:
(base) sbm_linux@sbmlinux:~/carla$ export UE4_ROOT=~/UnrealEngine_4.21 
(base) sbm_linux@sbmlinux:~/carla$ make relaunch
make: *** No rule to make target 'relaunch'.  Stop.
(base) sbm_linux@sbmlinux:~/carla$ make launch
Setup.sh: llvm-7.0 already installed.
Setup.sh: boost-1.69.0-c7 already installed.
cp: cannot create regular file 'boost-1.69.0-c7-install/include/boost/rational.hpp': No such file or directory
Util/BuildTools/Linux.mk:126: recipe for target 'setup' failed
make: *** [setup] Error 1

Other inputs:
OS: ubuntu 18.04.04 LTS
Some inputs: cmake version 3.15.2
Carla branch: master
tutorial: https://carla.readthedocs.io/en/latest/how_to_build_on_linux/

I tried switching to stable branch and rebuild but no effect.

any hint  ?

savinovava, 2020-02-24:
Have the same error with stable branch
[image]

bernatx, 2020-02-25:
Hi,
We have been testing on a Ubuntu 18.04 and we can make the Update.sh without a problem.
For the error message "gzip: stdin: not in gzip format" it seems that the download was failing and the file was corrupted. I have been able to download (the same version) and extract all the files.

For the boost library, it seems it could not extract the files either.

I propose to use the Carla.0.9.7 version, with Unreal Engine 4.22, because master branch is in constant development.

Regards

zkyxsq, 2020-02-28:
Hi  ，i have same question with you .

Setup.sh: Extracting boost for Python 2.

gzip: stdin: unexpected end of file
tar: Unexpected EOF in archive
tar: Unexpected EOF in archive
tar: Error is not recoverable: exiting now
Util/BuildTools/Linux.mk:126: recipe for target 'setup' failed
make: *** [setup] Error 2

mcemilg, 2020-02-28:
I was getting same issue while running `make launch` command. I deleted all corrupted boost files under `Build` folder and it worked properly.

closed 2020-05-06 by stale[bot]