carla-simulator/carla #2510: Problem with Waypoint`s methods

ManoelViktorovManev, 2020-02-25:
Hello! I use version 0.9.6 on Carla. I have problem with the methods in Carla.Waypoint class. In the Python API, I see methods like next_until_lane_end(self, distance), previous(self, distance) and previous_until_lane_start(self, distance), but when i tried to use them, it says ,,AttributeError: 'Waypoint' object has no attribute 'previous'". I know what this error means, but what is the reason that i can`t use these methods? Why the methods are describe in the Python API, but i can`t use them in my code.
Sorry if there are any mistakes in my English.

Axel1092, 2020-02-25:
Hi @ManoelViktorovManev! Looks like you are checking the documentation for the latest version of Carla which includes these methods, however, they were not implemented in version 0.9.6. Here you can find the documentation for the PythonAPI of version 0.9.6: https://github.com/carla-simulator/carla/blob/0.9.6/Docs/python_api.md. We are still working on providing the documentation for different version of Carla.

If you are Building Carla yourself through the repository in linux, you can build the documentation in your local machine running the following commands from the root of the repository:
```
pip install --user mkdocs pyyaml
make PythonAPI.docs
mkdocs serve
```

Then, open your web browser to this address: 127.0.0.1:8000 .

SunFishlike, 2020-02-25:
I tried the version 0.9.7.4,  but it still says

     'Waypoint' object has no attribute 'previous'

Axel1092, 2020-02-26:
Hello @SunFishlike! These functions are only included in the current in development master branch. You can find a packaged build of this version of Carla here: https://github.com/carla-simulator/carla/blob/master/Docs/download.md (the Nightly Build). Note that this is a development version with experimental changes.

closed 2020-05-03 by stale[bot]