carla-simulator/carla #2511: Segfault with Waypoint.next_until_lane_end method

whitesea1, 2020-02-25:
I think there is a bug with the new Waypoint.next_until_lane_end() method in the PythonAPI which results in a segfault if the returned list of waypoints would be zero (a call to the next method with the same distance results in a point in the next lane). 

An example case for testing:
In Town03, get the waypoint defined at:
road_id = 13
lane_id = -5
s = 0.0
with the get_waypoint_xodr() method of the map. This waypoint is in the roundabout for reference.

A call to next() with a distance of 2.0 results in a waypoint in the next lane segment, and a call to next_until_lane_end() with the same distance results in a segfault. 

My system:
Ubuntu 18.04
Python 3.6.10 (also tested with 3.5)
Carla 0.9.7
Unreal 4.22

manolotis, 2020-03-10:
Any updates or workarounds for this issue? I am facing the same problem and this method would be really convenient for me

whitesea1, 2020-03-10:
I ended up with a bit of a complicated workaround, not sure if it will be helpful but I can describe what I did.

First for context, my goal was to generate a static graph of the environment which could be used for some high level planning, i.e. a static set of waypoints with connections between them that capture the connections of waypoints within lanes as well as the connections between roads and lanes. 

1. I ended up parsing the opendrive file of the environment, and captured the connections between lane sections, as well as the lengths of each of the roads.
2. I then generated waypoints across the entire map with a small distance using the map.generate_waypoints() function and grouped the waypoints into lists by road_ids, section_ids, and lane_ids. Then I sorted each of those lists by the s-value of the waypoints (the direction of the sort depends on the sign of the lane_id).
3. At this point I have a bunch of lists representing sections of road, and the waypoints in each list are in order. From here I can just step through the lists by ((size of desired distance) / (size of small distance used to generate)) to get list of waypoints per road separated by the distance I want.
4. Now I just add connections between subsequent waypoints in each lane section, and then add connections to the beginning and end of each lane section by the connections I found in the opendrive format.

This is probably not what you are looking for, but it solved my problem.

To recreate the waypoint.next_until_lane_end() function, I think you should be able to call the waypoint.next() function in a loop until you get a waypoint with a new road_id or new lane_id. If you specifically need to include the very last possible waypoint in the lane, then I can think of two ways to handle that:

1. Go the the last waypoint you found that was in your lane, and then call next() with a very small distance (say .02 as that is the resolution that differentiates between waypoints) until the road_id or lane_id changes. Then just add the last waypoint you saw in your lane (or the first on in the new lane) to the end of your list of waypoints
2. Parse the Opendrive file for the town you are in to get the starting s and ending s of the lane you are in, and use the map.get_waypoint_xodr(road_id, lane_id, s) using either the starting s or ending s depending on which direction you are headed (sign of the lane_id) and the lane_id/road_id of the lane you care about.

 I saw that Carla 0.9.8 was release last night (woohoo!) but I am not sure if anything in that release may have fixed the bug with next_until_lane_end.

manolotis, 2020-03-11:
@whitesea1 Thanks a lot for the timely reply with your solution and suggestions! Indeed, this is not exactly what I was looking for, but it certainly gave me some ideas.
Just in case you are curious: I tried with Carla 0.9.8, and the problem with next_until_lane_end is still there.

nyukhalov, 2020-04-19:
I'm also facing the issue with Carla 0.9.8.
The below snippet reproduces the issue:

```python
import carla

client = carla.Client('localhost', 2000)
client.set_timeout(5.0)
client.load_world('Town02')

wp = client.get_world().get_map().get_waypoint(carla.Location(x=141.565140, y=240.929520, z=0.000000))
wp.next_until_lane_end(1)
```

The output:

```
Fatal Python error: Segmentation Fault
```

germanros1987, 2020-06-26:
@Axel1092 do you have any clue of what is going on? Could you please take a look at this?

Alby407, 2020-08-04:
I get a segmentation fault whenever I try to retrieve the map, i.e. `world.get_map()` in 0.9.8, as well as in 0.9.9.

Axel1092, 2020-08-20:
Sorry for the late response and thank you for your report we will investigate this issue. We will fix the behavior of `next_until_lane_end()` as soon as possible.

@Alby407 what map are you using? Can you provide us with the errors printed in the console (if any)?

closed 2020-11-08 by stale[bot]

tlqqidong, 2022-07-01:
@nyukhalov 
I got the same 'Segmentation Fault' with Carla 0.9.10.1. I have solved it by the following way:
`
if wp.lane_type != carla.LaneType.NONE:  
&nbsp;&nbsp;&nbsp;&nbsp; wp.next_until_lane_end(1)

`

adwiii, 2023-06-30:
I opened #6594 yesterday which appears to be the same underlying issue as this but manifesting in terms of `previous` instead of `next`. I am using a similar workaround as described [above](https://github.com/carla-simulator/carla/issues/2511#issuecomment-597230746) by manually re-creating the `previous_until_lane_start` functionality in Python and checking for self-loops explicitly.

353xiong, 2024-12-27:
Can someone please fix this bug?