carla-simulator/carla #2517: [0.9.7]WASD can't work in manual_control.py and after running ./CarlaUE4.sh

Leon-lhy, 2020-02-26:
I used teamviewer to run carla, In manual_control.py, 'wasd' can't control cars move but other keyboard control such 'c' can change weather and 'p' can switch into auto-mode, so I dont know how to solve this problem, How can i fix this problem? Thank you!

stevelaclasse, 2020-02-26:
i think it will not be easy to solve the problem, because teamviewer transmit the key you have pressed when you have release it, that is why 'c' and 'p', because they just need to be pressed one time in order to trigger the event, but for 'w' they have to be pressed in a continuous way for the car to move, as teamviewer transmit once you have release, the pygame key event just receive it as one 'w'. 

just try in a text editor to press for 5 seconds 'w' and then release it, to see what show up in the editor (using teamviewer).

Leon-lhy, 2020-02-26:
> i think it will not be easy to solve the problem, because teamviewer transmit the key you have pressed when you have release it, that is why 'c' and 'p', because they just need to be pressed one [...]

Thanks for your helping. I have successfully solved this problems by using other tools to connect remote server other than TeamViewer.

stevelaclasse, 2020-02-26:
which tools did you used ?

Leon-lhy, 2020-02-26:
> which tool did you used ?
> it may help others in the future.

AnyDesk

closed 2020-02-26 by Leon-lhy

stevelaclasse, 2020-02-27:
> i think it will not be easy to solve the problem, because teamviewer transmit the key you have pressed when you have release it, that is why 'c' and 'p', because they just need to be pressed one [...]

i tested it in the editor using Teamviewer and it work, but not on the Manual_control, so i was wrong about how Teamviewer transmit the data.

Leon-lhy, 2020-02-27:
> > i think it will not be easy to solve the problem, because teamviewer transmit the key you have pressed when you have release it, that is why 'c' and 'p', because they just need to be pressed one time in order to trigger the event, but for 'w' they have to be pressed in a continuous way for the car to move, as teamviewer transmit once you have release, the pygame key event just receive it as one 'w'.
> > just try in a text editor to press for 5 seconds 'w' and then release it, to see what show up in the editor (using teamviewer).
> 
> i tested it in the editor using Teamviewer and it works, but not on the Manual_control, so i was wrong about how Teamviewer transmit the data.

Although it works, this problem gives information that the way of transmission between anyDesk and Teamviewer is different. And I also found that it will shut down AnyDesk connection if I press 'w' in a long time, nearly 5s? So, stable connection is also a problem. Fortunately, keyboard can control in simulator finally.