carla-simulator/carla #2522: Confused with sensor.camera.depth and cc.Raw and cc.Depth

ocean1211, 2020-02-26:
From the website: https://carla.readthedocs.io/en/latest/cameras_and_sensors/, there are two types of depth image: the original one and the converted one (gray-scale).

And there is another instruction as:
The image codifies the depth in 3 channels of the RGB color space, from less to more significant bytes: R -> G -> B. The actual distance in meters can be decoded with
normalized = (R + G * 256 + B * 256 * 256) / (256 * 256 * 256 - 1)
in_meters = 1000 * normalized

My question is:
If I obtain the raw depth image( with image.convert(cc.Raw))  and calculate the depth according to the instruction above, will I get an image which is the same as the converted gray-scale image (with image.convert(cc.Depth)?

Thanks a lot for your answers!

cassfalg, 2020-02-26:
Images are typically 3 to 4 channel for colour (RGB + Alpha) or 1 to I guess 2 for greyscale (with alpha) with 8 or 16 bit integer values (1-2 byte) for each channel, values are then 0 to 255 or 0 to 65k. They can also be floating point, with 32 bit single precision for each value, then they are scaled from 0 to 1.

The CC.Depth converter is for visualizing depth maps, for which you will probably want a greyscale (or some other colour map) picture, probably on a log scale since that is better to look at.

If you want to use the depth for other things, like training a depth estimation network, or generating a point cloud, you want the depth in meters (or mm or km or whatever), so a floating point value (from 0 to infinity). For this you will want to use the decode formula. This can not be directly viewed by an image viewer (cause not scaled from 0 to 1). But this you can actually work with, not just look at.

ocean1211, 2020-02-27:
> Images are typically 3 to 4 channel for colour (RGB + Alpha) or 1 to I guess 2 for greyscale (with alpha) with 8 or 16 bit integer values (1-2 byte) for each channel, values are then 0 to 255 or 0 [...]

> Images are typically 3 to 4 channel for colour (RGB + Alpha) or 1 to I guess 2 for greyscale (with alpha) with 8 or 16 bit integer values (1-2 byte) for each channel, values are then 0 to 255 or 0 [...]

Thanks a lot for your reply!
So I will use the gray-scale image for visualization and the decoding value for the depth estimation. ;)

closed 2020-05-06 by stale[bot]