carla-simulator/carla #2524: Unable to connect to client object.

shishau, 2020-02-27:
I am a newbie linux.

The command in "Connecting and retrieving the world" does not run on https://carla.readthedocs.io/en/latest/python_api_tutorial/.

client = carla.Client('localhost', 2000)

In the localhost part
ip a
Substitute 127.0.0.1 from the above command.

In the 2000 part,
lsof -i
Substitute the number from the above command (eg 43334).

Execution result,
syntax error near unexpected token `('
(Because we are Japanese environment　予期しないトークン `(' 周辺に構文エラーがあります　It shows)

Is it a hierarchy problem that causes the command to be executed?
/carla
/carla/PythonAPI/axamples
Both had the same result.

I'm sorry, please tell me who you are.

environment
Ubuntu18.04
UnrealEngine4.22
CARLA0.9.7

shishau

amstrudy, 2020-03-04:
If the error you are getting is `syntax error near unexpected token`, then it's not a CARLA issue but probably just your Python syntax. What is the exact code you are running?

shishau, 2020-03-04:
Thank you for your reply!

client = carla.Client（ '127.0.0.1'、2000)

A syntax error is returned as a result of putting the above in the CUI.
What is the code in this case?

Sorry for not knowing much.

shishau

amstrudy, 2020-03-05:
You're almost there! The problem is the character after `'127.0.0.1'`, which Python is expecting to be a comma but looks more like a Japanese comma in your code? Anyways, the line should be:
```python
client = carla.Client('127.0.0.1', 2000)
```

closed 2020-05-13 by stale[bot]