carla-simulator/carla #2526: Make launch: 'boost/asio/buffer.hpp' not found

ScaramuzzinoGiovanna, 2020-02-27:
Hi,
i followed the tutorial: https://carla.readthedocs.io/en/latest/how_to_build_on_linux/ and i used:

_OS: ubuntu 18.04.04 LTS
cmake: version 3.10.2
Carla branch: 0.9.7
Unreal_engine: 4.22_

To clone the repository I used the following command:
`git clone --recursive --branch 0.9.7 https://github.com/carla-simulator/carla.git
`

After running 'make launch', the following error is shown:

```
Setup.sh: llvm-7.0 already installed.
Setup.sh: boost-1.69.0-c7 already installed.
Setup.sh: rpclib-v2.2.1_c2-c7 already installed.
Setup.sh: gtest-1.8.1-c7 already installed.
Setup.sh: recast-cdce4e-c7 already installed.
Setup.sh: CARLA version 0.9.7.
Setup.sh: Generating CMake configuration files.
Setup.sh: Success!
BuildLibCarla.sh: Building LibCarla "Server.Release" configuration.
[1/48] Building CXX object LibCarla/cmake/server/C...ce/carla/opendrive/parser/GeoReferenceParser.cpp.o
FAILED: LibCarla/cmake/server/CMakeFiles/carla_server.dir/__/__/source/carla/opendrive/parser/GeoReferenceParser.cpp.o 
/usr/bin/clang++-7  -DASIO_NO_EXCEPTIONS -DBOOST_ERROR_CODE_HEADER_ONLY -DBOOST_NO_EXCEPTIONS -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true -DLIBCARLA_NO_EXCEPTIONS -DLIBCARLA_TEST_CONTENT_FOLDER=\"/home/giovanna/carla/Build/test-content\" -DPUGIXML_NO_EXCEPTIONS -I../../LibCarla/cmake/../source -I../../LibCarla/cmake/../source/third-party -isystem ../boost-1.69.0-c7-install/include -isystem ../rpclib-v2.2.1_c2-c7-libcxx-install/include -std=c++14 -pthread -fPIC -Werror -Wall -Wextra -Wpedantic -Wdeprecated -Wshadow -Wuninitialized -Wunreachable-code -Wpessimizing-move -Wold-style-cast -Wnull-dereference -Wduplicate-enum -Wnon-virtual-dtor -Wheader-hygiene -Wconversion -Wfloat-overflow-conversion -stdlib=libc++ -isystem /home/giovanna/carla/Build/llvm-7.0-install/include/c++/v1 -fno-exceptions -fno-rtti -std=c++14 -pthread -fPIC -Werror -Wall -Wextra -Wpedantic -Wdeprecated -Wshadow -Wuninitialized -Wunreachable-code -Wpessimizing-move -Wold-style-cast -Wnull-dereference -Wduplicate-enum -Wnon-virtual-dtor -Wheader-hygiene -Wconversion -Wfloat-overflow-conversion -stdlib=libc++ -isystem /home/giovanna/carla/Build/llvm-7.0-install/include/c++/v1 -fno-exceptions -fno-rtti   -O3 -DNDEBUG -MD -MT LibCarla/cmake/server/CMakeFiles/carla_server.dir/__/__/source/carla/opendrive/parser/GeoReferenceParser.cpp.o -MF LibCarla/cmake/server/CMakeFiles/carla_server.dir/__/__/source/carla/opendrive/parser/GeoReferenceParser.cpp.o.d -o LibCarla/cmake/server/CMakeFiles/carla_server.dir/__/__/source/carla/opendrive/parser/GeoReferenceParser.cpp.o -c ../../LibCarla/source/carla/opendrive/parser/GeoReferenceParser.cpp
In file included from ../../LibCarla/source/carla/opendrive/parser/GeoReferenceParser.cpp:10:
../../LibCarla/cmake/../source/carla/StringUtil.h:9:10: fatal error: 'boost/algorithm/string.hpp' file not found
#include <boost/algorithm/string.hpp>
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
[2/48] Building CXX object LibCarla/cmake/server/C...rver.dir/__/__/source/carla/geom/GeoLocation.cpp.o
FAILED: LibCarla/cmake/server/CMakeFiles/carla_server.dir/__/__/source/carla/geom/GeoLocation.cpp.o 
/usr/bin/clang++-7  -DASIO_NO_EXCEPTIONS -DBOOST_ERROR_CODE_HEADER_ONLY -DBOOST_NO_EXCEPTIONS -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true -DLIBCARLA_NO_EXCEPTIONS -DLIBCARLA_TEST_CONTENT_FOLDER=\"/home/giovanna/carla/Build/test-content\" -DPUGIXML_NO_EXCEPTIONS -I../../LibCarla/cmake/../source -I../../LibCarla/cmake/../source/third-party -isystem ../boost-1.69.0-c7-install/include -isystem ../rpclib-v2.2.1_c2-c7-libcxx-install/include -std=c++14 -pthread -fPIC -Werror -Wall -Wextra -Wpedantic -Wdeprecated -Wshadow -Wuninitialized -Wunreachable-code -Wpessimizing-move -Wold-style-cast -Wnull-dereference -Wduplicate-enum -Wnon-virtual-dtor -Wheader-hygiene -Wconversion -Wfloat-overflow-conversion -stdlib=libc++ -isystem /home/giovanna/carla/Build/llvm-7.0-install/include/c++/v1 -fno-exceptions -fno-rtti -std=c++14 -pthread -fPIC -Werror -Wall -Wextra -Wpedantic -Wdeprecated -Wshadow -Wuninitialized -Wunreachable-code -Wpessimizing-move -Wold-style-cast -Wnull-dereference -Wduplicate-enum -Wnon-virtual-dtor -Wheader-hygiene -Wconversion -Wfloat-overflow-conversion -stdlib=libc++ -isystem /home/giovanna/carla/Build/llvm-7.0-install/include/c++/v1 -fno-exceptions -fno-rtti   -O3 -DNDEBUG -MD -MT LibCarla/cmake/server/CMakeFiles/carla_server.dir/__/__/source/carla/geom/GeoLocation.cpp.o -MF LibCarla/cmake/server/CMakeFiles/carla_server.dir/__/__/source/carla/geom/GeoLocation.cpp.o.d -o LibCarla/cmake/server/CMakeFiles/carla_server.dir/__/__/source/carla/geom/GeoLocation.cpp.o -c ../../LibCarla/source/carla/geom/GeoLocation.cpp
In file included from ../../LibCarla/source/carla/geom/GeoLocation.cpp:7:
In file included from ../../LibCarla/cmake/../source/carla/geom/GeoLocation.h:9:
In file included from ../../LibCarla/cmake/../source/carla/MsgPack.h:9:
../../LibCarla/cmake/../source/carla/Buffer.h:13:10: fatal error: 'boost/asio/buffer.hpp' file not found
#include <boost/asio/buffer.hpp>
         ^~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
[3/48] Building CXX object LibCarla/cmake/server/C...__/__/source/carla/opendrive/OpenDriveParser.cpp.o
FAILED: LibCarla/cmake/server/CMakeFiles/carla_server.dir/__/__/source/carla/opendrive/OpenDriveParser.cpp.o 
/usr/bin/clang++-7  -DASIO_NO_EXCEPTIONS -DBOOST_ERROR_CODE_HEADER_ONLY -DBOOST_NO_EXCEPTIONS -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true -DLIBCARLA_NO_EXCEPTIONS -DLIBCARLA_TEST_CONTENT_FOLDER=\"/home/giovanna/carla/Build/test-content\" -DPUGIXML_NO_EXCEPTIONS -I../../LibCarla/cmake/../source -I../../LibCarla/cmake/../source/third-party -isystem ../boost-1.69.0-c7-install/include -isystem ../rpclib-v2.2.1_c2-c7-libcxx-install/include -std=c++14 -pthread -fPIC -Werror -Wall -Wextra -Wpedantic -Wdeprecated -Wshadow -Wuninitialized -Wunreachable-code -Wpessimizing-move -Wold-style-cast -Wnull-dereference -Wduplicate-enum -Wnon-virtual-dtor -Wheader-hygiene -Wconversion -Wfloat-overflow-conversion -stdlib=libc++ -isystem /home/giovanna/carla/Build/llvm-7.0-install/include/c++/v1 -fno-exceptions -fno-rtti -std=c++14 -pthread -fPIC -Werror -Wall -Wextra -Wpedantic -Wdeprecated -Wshadow -Wuninitialized -Wunreachable-code -Wpessimizing-move -Wold-style-cast -Wnull-dereference -Wduplicate-enum -Wnon-virtual-dtor -Wheader-hygiene -Wconversion -Wfloat-overflow-conversion -stdlib=libc++ -isystem /home/giovanna/carla/Build/llvm-7.0-install/include/c++/v1 -fno-exceptions -fno-rtti   -O3 -DNDEBUG -MD -MT LibCarla/cmake/server/CMakeFiles/carla_server.dir/__/__/source/carla/opendrive/OpenDriveParser.cpp.o -MF LibCarla/cmake/server/CMakeFiles/carla_server.dir/__/__/source/carla/opendrive/OpenDriveParser.cpp.o.d -o LibCarla/cmake/server/CMakeFiles/carla_server.dir/__/__/source/carla/opendrive/OpenDriveParser.cpp.o -c ../../LibCarla/source/carla/opendrive/OpenDriveParser.cpp
In file included from ../../LibCarla/source/carla/opendrive/OpenDriveParser.cpp:7:
In file included from ../../LibCarla/cmake/../source/carla/opendrive/OpenDriveParser.h:9:
In file included from ../../LibCarla/cmake/../source/carla/road/Map.h:10:
In file included from ../../LibCarla/cmake/../source/carla/geom/Transform.h:9:
In file included from ../../LibCarla/cmake/../source/carla/MsgPack.h:9:
../../LibCarla/cmake/../source/carla/Buffer.h:13:10: fatal error: 'boost/asio/buffer.hpp' file not found
#include <boost/asio/buffer.hpp>
         ^~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
[4/48] Building CXX object LibCarla/cmake/server/C.../__/source/carla/opendrive/parser/LaneParser.cpp.o
FAILED: LibCarla/cmake/server/CMakeFiles/carla_server.dir/__/__/source/carla/opendrive/parser/LaneParser.cpp.o 
/usr/bin/clang++-7  -DASIO_NO_EXCEPTIONS -DBOOST_ERROR_CODE_HEADER_ONLY -DBOOST_NO_EXCEPTIONS -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true -DLIBCARLA_NO_EXCEPTIONS -DLIBCARLA_TEST_CONTENT_FOLDER=\"/home/giovanna/carla/Build/test-content\" -DPUGIXML_NO_EXCEPTIONS -I../../LibCarla/cmake/../source -I../../LibCarla/cmake/../source/third-party -isystem ../boost-1.69.0-c7-install/include -isystem ../rpclib-v2.2.1_c2-c7-libcxx-install/include -std=c++14 -pthread -fPIC -Werror -Wall -Wextra -Wpedantic -Wdeprecated -Wshadow -Wuninitialized -Wunreachable-code -Wpessimizing-move -Wold-style-cast -Wnull-dereference -Wduplicate-enum -Wnon-virtual-dtor -Wheader-hygiene -Wconversion -Wfloat-overflow-conversion -stdlib=libc++ -isystem /home/giovanna/carla/Build/llvm-7.0-install/include/c++/v1 -fno-exceptions -fno-rtti -st [... cut ...]
In file included from ../../LibCarla/cmake/../source/carla/geom/Vector3D.h:9:
In file included from ../../LibCarla/cmake/../source/carla/MsgPack.h:9:
../../LibCarla/cmake/../source/carla/Buffer.h:13:10: fatal error: 'boost/asio/buffer.hpp' file not found
#include <boost/asio/buffer.hpp>
         ^~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
ninja: build stopped: subcommand failed.
Util/BuildTools/Linux.mk:111: recipe for target 'LibCarla.server.release' failed
make: *** [LibCarla.server.release] Error 1
```
Does anyone know how to solve this problem?

mcemilg, 2020-02-28:
I encountered with same issue with same system specs. Also there is another one: #2501

mcemilg, 2020-02-28:
I solved my issue, maybe it will help you too. The problem is that the boost library did not installed properly. Check boost folders under the `Build` folder. If `boost-1.72.0-c7-source` and `boost-1.72.0-c7-install` folders are exist, copy all files and folders under source folder to install folder. If it's not solved the issue try to remove all boost related files and folders, rerun the `make launch` command.

ScaramuzzinoGiovanna, 2020-03-02:
Unfortunately this didn't work out for me. Thanks anyway

mcfly50, 2020-03-03:
@mcemilg  - removing the boost source folder, install folder and tar.gz from the /Build folder and then doing "make launch" again worked!

Trying to copy the files and folders into install folder did not work.

Thanks

ScaramuzzinoGiovanna, 2020-03-03:
It worked by removing the three files and relaunching the _make launch._

Thank you very much.

closed 2020-03-03 by ScaramuzzinoGiovanna

vishal-246, 2020-05-27:
Hi ,

i tried the removing 'boost-1.72.0-c8-install'  below is the log:

```
$ make LibCarla
Setup.sh: llvm-8.0 already installed.
Setup.sh: boost-1.72.0-c8 already installed.
Setup.sh: rpclib-v2.2.1_c2-c8 already installed.
Setup.sh: gtest-1.8.1-c8 already installed.
Setup.sh: recast-cdce4e-c8 already installed.
Setup.sh: CARLA version 0.9.9-58-g4dc4cb81-dirty.
Setup.sh: Generating CMake configuration files.
Setup.sh: Success!
BuildLibCarla.sh: Building LibCarla "Server.Release" configuration.
ninja: no work to do.
[0/1] Install the project...
-- Install configuration: "Server"
BuildLibCarla.sh: Success!
BuildLibCarla.sh: Building LibCarla "Client.Release" configuration.
-- The C compiler identification is Clang 8.0.1
-- The CXX compiler identification is Clang 8.0.1
-- Could not determine Eclipse version, assuming at least 3.6 (Helios). Adjust CMAKE_ECLIPSE_VERSION if this is wrong.
-- Check for working C compiler: /usr/bin/clang-8
-- Check for working C compiler: /usr/bin/clang-8 -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/clang++-8
-- Check for working CXX compiler: /usr/bin/clang++-8 -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Build debug:   OFF
-- Build release: ON
-- Build test:    ON
-- Configuring done
CMake Warning in CMakeLists.txt:
  The build directory is a subdirectory of the source directory.

  This is not supported well by Eclipse.  It is strongly recommended to use a
  build directory which is a sibling of the source directory.

-- Generating done
-- Build files have been written to: /carla/carla/Build/libcarla-client-build.release
[5/118] Building CXX object LibCarla/cmake/client/...a_client.dir/__/__/source/carla/client/Actor.cpp.o
FAILED: LibCarla/cmake/client/CMakeFiles/carla_client.dir/__/__/source/carla/client/Actor.cpp.o 
/usr/bin/clang++-8  -DBOOST_ERROR_CODE_HEADER_ONLY -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true -DLIBCARLA_TEST_CONTENT_FOLDER=\"/carla/carla/Build/test-content\" -I../../LibCarla/cmake/../source -I../../LibCarla/cmake/../source/third-party -isystem ../boost-1.72.0-c8-install/include -isystem ../rpclib-v2.2.1_c2-c8-libstdcxx-install/include -isystem ../recast-cdce4e-c8-install/include -std=c++14 -pthread -fPIC -Werror -Wall -Wextra -Wpedantic -Wdeprecated -Wshadow -Wuninitialized -Wunreachable-code -Wpessimizing-move -Wold-style-cast -Wnull-dereference -Wduplicate-enum -Wnon-virtual-dtor -Wheader-hygiene -Wconversion -Wfloat-overflow-conversion -std=c++14 -pthread -fPIC -Werror -Wall -Wextra -Wpedantic -Wdeprecated -Wshadow -Wuninitialized -Wunreachable-code -Wpessimizing-move -Wold-style-cast -Wnull-dereference -Wduplicate-enum -Wnon-virtual-dtor -Wheader-hygiene -Wconversion -Wfloat-overflow-conversion    -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -Wdate-time -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -std=c++14 -Wno-missing-braces -DBOOST_ERROR_CODE_HEADER_ONLY -MD -MT LibCarla/cmake/client/CMakeFiles/carla_client.dir/__/__/source/carla/client/Actor.cpp.o -MF LibCarla/cmake/client/CMakeFiles/carla_client.dir/__/__/source/carla/client/Actor.cpp.o.d -o LibCarla/cmake/client/CMakeFiles/carla_client.dir/__/__/source/carla/client/Actor.cpp.o -c ../../LibCarla/source/carla/client/Actor.cpp
In file included from ../../LibCarla/source/carla/client/Actor.cpp:7:
In file included from ../../LibCarla/cmake/../source/carla/client/Actor.h:11:
In file included from ../../LibCarla/cmake/../source/carla/client/detail/ActorState.h:10:
In file included from ../../LibCarla/cmake/../source/carla/client/World.h:13:
In file included from ../../LibCarla/cmake/../source/carla/client/LightManager.h:17:
In file included from ../../LibCarla/cmake/../source/carla/client/detail/Episode.h:17:
In file included from ../../LibCarla/cmake/../source/carla/client/detail/WalkerNavigation.h:10:
../../LibCarla/cmake/../source/carla/nav/Navigation.h:16:10: fatal error: 'recast/Recast.h' file not found
#include <recast/Recast.h>
         ^~~~~~~~~~~~~~~~~
1 error generated.
[8/118] Building CXX object LibCarla/cmake/client/...nt.dir/__/__/source/carla/client/DebugHelper.cpp.o
FAILED: LibCarla/cmake/client/CMakeFiles/carla_client.dir/__/__/source/carla/client/DebugHelper.cpp.o 
/usr/bin/clang++-8  -DBOOST_ERROR_CODE_HEADER_ONLY -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true -DLIBCARLA_TEST_CONTENT_FOLDER=\"/carla/carla/Build/test-content\" -I../../LibCarla/cmake/../source -I../../LibCarla/cmake/../source/third-party -isystem ../boost-1.72.0-c8-install/include -isystem ../rpclib-v2.2.1_c2-c8-libstdcxx-install/include -isystem ../recast-cdce4e-c8-install/include -std=c++14 -pthread -fPIC -Werror -Wall -Wextra -Wpedantic -Wdeprecated -Wshadow -Wuninitialized -Wunreachable-code -Wpessimizing-move -Wold-style-cast -Wnull-dereference -Wduplicate-enum -Wnon-virtual-dtor -Wheader-hygiene -Wconversion -Wfloat-overflow-conversion -std=c++14 -pthread -fPIC -Werror -Wall -Wextra -Wpedantic -Wdeprecated -Wshadow -Wuninitialized -Wunreachable-code -Wpessimizing-move -Wold-style-cast -Wnull-dereference -Wduplicate-enum -Wnon-virtual-dtor -Wheader-hygiene -Wconversion -Wfloat-overflow-conversion    -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -Wdate-time -D_FO [... cut ...]
[10/118] Building CXX object LibCarla/cmake/client...dir/__/__/source/carla/client/ActorBlueprint.cpp.o
ninja: build stopped: subcommand failed.
Util/BuildTools/Linux.mk:126: recipe for target 'LibCarla.client.release' failed
make: *** [LibCarla.client.release] Error 1
$
````

From this i can udesrstand that recast is making problem here. but the path specified for recast in `/carla/build/CMakeLists.txt.in` is already there in `/carla/build/recast-cdce4e-c8-install` folder

So please help me solve this issue.

Thank you.

vishal-246, 2020-05-27:
issue is solved by removing the recast folders and making it again.

thank you for the solution.

TrungKhoaLe, 2020-06-21:
> @mcemilg - removing the boost source folder, install folder and tar.gz from the /Build folder and then doing "make launch" again worked!
> 
> Trying to copy the files and folders into install folder did not work.
> 
> Thanks
Worked for me too