carla-simulator/carla #2533: can't connect to the Boost server

dhruvildarji, 2020-02-29:
https://dl.bintray.com/boostorg/release/1.72.0/source/boost_1_72_0.zip

This link does not work with any device. 

Any help?

mcfly50, 2020-03-02:
Same problem. Carla, master branch. Update.sh completes successfully.

'make launch' results in the following:

Setup.sh: Retrieving boost.
--2020-03-01 21:51:22--  https://dl.bintray.com/boostorg/release/1.72.0/source/boost_1_72_0.tar.gz
Resolving dl.bintray.com (dl.bintray.com)... 35.166.139.238, 52.39.25.243
Connecting to dl.bintray.com (dl.bintray.com)|35.166.139.238|:443... connected.
HTTP request sent, awaiting response... 403 Forbidden
2020-03-01 21:51:23 ERROR 403: Forbidden.

Util/BuildTools/Linux.mk:126: recipe for target 'setup' failed
make: *** [setup] Error 8

kzykogo0415, 2020-03-02:
I also face same problem. Is this error has occur recently?

kzykogo0415, 2020-03-02:
I discover this site. Down load this and save the folder  carla\Build
https://sourceforge.net/projects/boost/files/boost/1.72.0/

mcfly50, 2020-03-03:
This issue with boost server indicating 'forbidden' seems to be fixed. Was able to connect to boost server yesterday.

TheGermanKid, 2020-03-28:
Has anybody found a solution/workaround to this problem or is this something on bintrays side? I just had the exact same issues and can't get around it!

Saytas, 2020-03-28:
[image]
I am also having the same issue. Did you guys solve the issue?

savinovava, 2020-03-28:
Have the same. 
[image]

savinovava, 2020-03-28:
> I discover this site. Down load this and save the folder carla\Build
> https://sourceforge.net/projects/boost/files/boost/1.72.0/

It doesn't help, not in tar.gz, not in unzip format

revikiran1991, 2020-03-28:
https://github.com/carla-simulator/carla/issues/2664#issuecomment-605438816 

temporary fix.

closed 2020-06-06 by stale[bot]

fliedroc, 2021-11-23:
> I discover this site. Down load this and save the folder carla\Build https://sourceforge.net/projects/boost/files/boost/1.72.0/

it work well for me, thanks