carla-simulator/carla #2538: Simulating collisions at intersections

Armandomiguel, 2020-03-01:
Hello everyone,

I'm triying to recreate accident scenarios using Carla and I would like to know which is the best way of building such scenarios.

I did spawned vehicles at an specific position (see the attached image) and turn autopilot on for them to drive into the intersection, but I've found two issues:

1. I cannot specify the route they are going to take (I tried using the function set_destination on BasicAgent but it dind't work). Is there a way of using the generated waypoints for telling the vehicles where the should drive to? or any other way of pecifiying the route?

2. The autopilot takes traffic regulations into consideration and avoids collisions, but I would like to let the vehicles approach the intersection "as aggressively as possible" (i.e. ignoring traffic regulations and the other vehicle). Is there a way of doing that?

I'd really appreciate your help on this.

PS. I'm running Carla 0.9.7 on Ubuntu 18.04

[image]

Armandomiguel, 2020-03-01:
Hello again!

I did found the solution to my problem. Here the answers (only for sharing purposes):

For 1.: the BasicAgent allows me to set the destination by using the function set_destination. 
    vehicle_agent = BasicAgent(vehicle, target_speed=50)
    vehicle_agent.set_destination([130, 326, 2.5])
The agent was not working for me because I didn't run the steps. This is done as follows:
        vehicle_control = vehicle_agent.run_step()
        vehicle.apply_control(vehicle_control) 

For 2.: the basic_agent.py file (which contains the BasicAgent class) includes the mechanism for taking traffic signals and other vehicles in the world into consideration. By commenting this parts of the code I could make the vehicles drive aggressively.
Additionally, a method for "ignoring" traffic lights (e.g. when collisions are desired) is to get all of the traffic lights on the world and set them permanently to green. 
        for tl in world.get_actors().filter('traffic.traffic_light*'):
        tl.set_state(carla.TrafficLightState.Green)
        tl.freeze(True)

I hope this information could be useful for people who wants to do the same.

And if you know a better method for accomplishing this, please tell me.

Best regards.

closed 2020-03-01 by Armandomiguel

Sibozhu, 2022-01-19:
hi @armandomiguel thanks for this post! I'm actually trying to do the same. Besides the collision of cars, is that possible we also simulator other normal traffic accidents like pedestrian+bike, pedestrian+car?

Best,
Sibo

armandomiguel, 2022-01-24:
Hi @Sibozhu, now it is indeed possible to simulate traffic accidents involving pedestrians.
In order to control the pedestrians, you can simply use the WalkerControl (https://carla.readthedocs.io/en/latest/core_actors/#walkers). Then you can simulate the collisions at intersections by spawning the vehicles and pedestrians near the intersection and letting them drive/walk into it.

Sibozhu, 2022-01-24:
Hi @armandomiguel thanks for the reply! I just realized the walker control as well! By any chance do you still have the original code of simulating collisions that I can take for reference?

Thanks a lot

armandomiguel, 2022-01-25:
@Sibozhu Unfortunately, I'm not allowed to share the original code.
But you can drop me an email at armando.miguel.garcia[at]gmail.com if you have any specific questions about how to build such a simulation.