carla-simulator/carla #2539: Actors not moving in synchronous mode

rueeggmi, 2020-03-02:
I use two clients, one for data collection, the other for spawning actors (spawn_npc.py). The first runs in synchronous mode to make sure no data is lost. But I have the problem that all actors from the second client do not move when I see them through the cameras of the first client. When running an asynchronous script (e.g. manual_control.py) all of the spawned pedestrians can be seen moving. Does anyone know how to solve this issue? Thanks a lot in advance!
(CARLA version 0.9.6 running on Ubuntu 18.04)

Schuck84, 2020-03-15:
I have the same issue on Ubuntu 16.04 using CARLA 0.9.8. I used in Python world.tick() but my actors are not moving when I set autopilot(True). Maybe a problem with the TrafficManager module?

Schuck84, 2020-03-15:
I got it :)
There is a "hidden" function "set_synchronous_mode" for the TrafficManager that is not documented in the Python API. I got it with the following code working. 
`tm = client.get_trafficmanager()
    TrafficManager.set_synchronous_mode(tm, True)`

Rajat-Mehta, 2020-04-14:
@Schuck84 where should I add these two lines of codes? In spawn_npc.py or synchronous_mode.py? I tried to add in both of them but it doesn't work for me.

rueeggmi, 2020-04-14:
I ended up using the spawn_npc.py file as a thread in the other client I was using to make it work. This issue might also help you, as it is somewhat related: https://github.com/carla-simulator/scenario_runner/issues/495

closed 2020-04-14 by rueeggmi