carla-simulator/carla #2542: Car following project

hamdii02, 2020-03-02:
Hello,
I am working on a project and i want to simulate a leader car in front of my car and i want to know how i can do this with carla
Thanks

hita-bm, 2020-03-02:
Hi @hamdii02 
Better to take a look at the [scenario runner](https://github.com/carla-simulator/scenario_runner) project. It has an example called [FollowLeadingVehicle](https://carla-scenariorunner.readthedocs.io/en/latest/getting_started/index.html#running-the-follow-vehicle-example). You can get how to run, from its [docs](https://carla-scenariorunner.readthedocs.io/en/latest/).
Good luck.

hamdii02, 2020-03-02:
@hita-bm ok thanks for responding and i will take a look on it

hita-bm, 2020-03-02:
@hamdii02 You're welcome. I recommend using Carla 0.9.6 and Scenario Runner 0.9.6 because the latest version (0.9.7) still has some bugs and needs to be completed. Trust me!

hamdii02, 2020-03-02:
@hita-bm  that's exactly what am using  and i have already tried the 0.9.7 version and i've had a lot of bugs but the 9.6.0 is good for now

closed 2020-05-09 by stale[bot]