carla-simulator/carla #2544: _pid_control accounts for time discretization twice (not once)

rowanmcallister, 2020-03-03:
It appears `_pid_control` incorrectly accounts for `self._dt` *twice* (not once), for both the "D" and "I" terms, and is therefore sensitive to the value of the time discretization (bad!). For example, currently the "D" term is:
`self._K_D * _de / self._dt == `
`self._K_D * (self._e_buffer[-1] - self._e_buffer[-2]) / self._dt / self._dt`

But it should be:
`self._K_D * (self._e_buffer[-1] - self._e_buffer[-2]) / self._dt`
by either removing `self._dt` from lines 117-118, or line 123.

https://github.com/carla-simulator/carla/blob/master/PythonAPI/carla/agents/navigation/controller.py#L117-L123

rowanmcallister, 2020-03-03:
And same thing for `PIDLateralController` too.

jackbart94, 2020-03-03:
Hi, thanks for pointing this out!

Actually I noticed the same thing and applied the change to another agent (not yet in master unfortunately, but a PR is open) however this would call for a PID re-tuning for both the Basic and Roaming agent, which can be tedious. If you are willing to, you can open a PR yourself with the changes, it would help us a lot!

ZhangYouRong, 2020-03-05:
It seems that this PID control function has static error because the KI term has an maximum queue length of 30 `self._e_buffer = deque(maxlen=30)`. How can I solve this problem?

PR #2550 (2020-03-16): Make PID controller insensitive to time-discretization
#### Description

Updated PID controller, to make it insensitive to chosen value of time discretization.

Fixes #2544

#### Where has this been tested?

  * **Platform(s):** Ubuntu 16.04
  * **Python version(s):** 3.6.10
  * **Unreal Engine version(s):** 4.18

#### Possible Drawbacks

If people have already tuned PID controllers on non-default time discretizations (non 0.05 values), they will need to re-tune their PID `K_I` and `K_D` terms by multiplying and diving by a factor of `self._dt / 0.05` respectively.

---
This change is [[image] [... cut]
closed 2020-03-16 by bernatx