carla-simulator/carla #2546: Cmake error while building LibCarla

incealper, 2020-03-03:
Hi Everyone !
When I try to build Carla, I encounter with this error yet VS 2017 has been already installed on my computer with Desktop Development C++ and 8.1 SDK. Can you help me out? I am really struggling with this.

make LibCarla and other make commands result like this:

**

j4l-Shvn, 2020-03-06:
Any response of this yet?

barasm-hita, 2020-03-09:
Hi @incealper 
If you already have installed **make** and **Cmake** on your Windows and added them to the **system path**, one thing that causes this error is using Command Prompt or PowerShell in Windows.
If you are using one of these, it's a mistake. You have to use "**x64 Native Tools Command Prompt for VS 2017**" from the apps menu. I think it will solve your problem.

mixeur5, 2020-03-12:
Hello @barasm-hita, @incealper 
I have the same problem. I have set up my environment following the build on windows documentation (visual studio 2017, sdk 8.1, Cmake...). At the beginning the Setup.bat script launched by the make command couldn't find cmake.

My Cmake installation was located in C:/Program Files, and as soon as I placed it in a different folder of my PATH, the bat scripts were able to find it.
Therefore there seems to be a problem with the way nmake handles the environment PATH.
I think it may be an issue caused by the spaces in the path name.

I am now facing a quite similar problem. During the boost installation, the bootstrap.bat script isn't able to find the vswhere.exe.

` -[install_boost]: Retrieving boost.
    -[install_boost]: Extracting boost from "boost_1_69_0.zip", this can take a while...
    -[install_boost]: Removing "boost_1_69_0.zip"
    -[install_boost]: Generating build...
Building Boost.Build engine
'vswhere' n’est pas reconnu en tant que commande interne
ou externe, un programme exécutable ou un fichier de commandes.
'vswhere' n’est pas reconnu en tant que commande interne
ou externe, un programme exécutable ou un fichier de commandes.`

I'll keep you posted on my findings, but any help would be greatly apreciated.

barasm-hita, 2020-03-13:
Hi @mixeur5 
Actually I'm not sure about why does this error appear in your boost build. But if your problem is 'vswhere' missing, you can install it manually. [Here](https://github.com/microsoft/vswhere/wiki/Installing) is shown how to install it.

mixeur5, 2020-03-18:
Thanks for your reply @barasm-hita. 
I have installed vswhere manually. However I had also another problem with the b2.exe build in the install_boost.bat. I had an error message "link: extra operand"
After searching on the internet, I found that thisr is a MinGW32 error. It was caused by the fact that MinGW32 was overriding the x64 visual studio cmd build tools, and was using its own link.exe.
Launching the b2 build in a standalone script (in place of using make launch) solved the problem. 
As I said before, launching batch files from makefiles (as it is done in the current Carla build pipeline) seems to cause MinGW32 to override the visual studio cmd prompt.
Anyway, I managed to build Carla successfully, but I guess the Carla build pipeline is not so robust and should be revised.

closed 2020-05-24 by stale[bot]

Kenneth71-P, 2022-01-04:
[image]
[image]
I had the same problem. How should i solve the problem?

xiaoqingnianbudadun, 2022-01-04:
u need upgrade cmake to version3.9.0 or higher

Kenneth71-P, 2022-01-04:
> 

But I use Cmake version 3.22.0. So I need to make sure my computer can find cmake path

Kenneth71-P, 2022-01-04:
> 
> 
> u need upgrade cmake to version3.9.0 or higher 
> 
> 
> 
> ---Original--- 
> From: ***@***.***&gt; 
> Date: Tue, Jan 4, 2022 13:48 PM 
> To: ***@***.***&gt;; 
> Cc: ***@***.***&gt;; 
But I use Cmake version 3.22.0 and I need to make sure my window os computer can find the cmake path Can you explain more

martino92, 2022-01-21:
> Hi @incealper If you already have installed **make** and **Cmake** on your Windows and added them to the **system path**, one thing that causes this error is using Command Prompt or PowerShell in [...]

It doesn't help

Kenneth71-P, 2022-01-21:
> 

Okay thank you

Kenneth71-P, 2022-01-21:
Okay thank you

vibhoraggarwal, 2022-06-24:
> [image] [image] I had the same problem. How should i solve the problem?

@Kenneth71-P I am haivng the same problem even though I have a cmake v3.20.5. Did you solve the problem already?

Kenneth71-P, 2022-06-26:
Yes I solved the problem

Kenneth71-P, 2022-06-26:
> > [image] [image] 나는 같은 문제가 있었다. 문제를 어떻게 해결해야합니까?
> 
> @Kenneth71-P 나는 cmake v3.20.5를 가지고 있음에도 불구하고 같은 문제가 있습니다. 이미 문제를 해결 했습니까?

But I forgot how to solve the problem. Sorry about that.

shott92, 2022-07-16:
Also having this problem, I am using X64 native tools, as admin, I have cmake version 3.23.2, make version 3.81 and VS version 17 (2022), I've ran the build guide 3 times now and used tips from FAQs but keep getting stuck at this build point, I was manually installing the build dependancies to get zlib and libpng, to work, but this hasn't helpped for rpclib.
[image]

tonmoay, 2023-01-16:
I worked around the issue for two days. It turns out as a simple 'Git' issue. 'install_rpclib.bat' uses 'Git' to pull the rpclib repo.

`echo git clone -b "%RPC_VERSION%" https://github.com/carla-simulator/rpclib.git "%RPC_SRC_DIR:~0,-1%"
 call git clone -b "%RPC_VERSION%" https://github.com/carla-simulator/rpclib.git "%RPC_SRC_DIR:~0,-1%"`

pranavambat, 2023-06-21:
> 

Where did you run the command?

Emilyhelin, 2024-07-05:
> Also having this problem, I am using X64 native tools, as admin, I have cmake version 3.23.2, make version 3.81 and VS version 17 (2022), I've ran the build guide 3 times now and used tips from FAQs but keep getting stuck at this build point, I was manually installing the build dependancies to get zlib and libpng, to work, but this hasn't helpped for rpclib. [image]
> 
> EDIT/UPDATE - resolved by installing Visual Stuido 2019 and running from the X64 Native Tool cmd for Visual studio 2019. Think the make script needs updating to work with VS22

I solved it by substuting make pythonAPI for make pythonAPI GENERATOR="Visual Studio 17 2022"

ajay1606, 2025-02-07:
@Emilyhelin and @ARRETdaniel Thanks for you responses, I have followed the same as you mentioned. But getting error same as bellow,

[image]

Any suggestions please ?