carla-simulator/carla #2548: makelaunch carla error Util/BuildTools/Linux.mk:7: recipe for target 'launch' failed make: *** [launch] Error 2

abdallahaymaan, 2020-03-03:
Carla was successfully built but the launch phase has an error that says: Util/BuildTools/Linux.mk:7: recipe for target 'launch' failed
make: *** [launch] Error 2

here is what was stated before
Setup.sh: llvm-7.0 already installed.
Setup.sh: boost-1.72.0-c7 already installed.
Setup.sh: rpclib-v2.2.1_c2-c7 already installed.
Setup.sh: gtest-1.8.1-c7 already installed.
Setup.sh: recast-cdce4e-c7 already installed.
Setup.sh: CARLA version 0.9.7-153-g85b19253.
Setup.sh: Generating CMake configuration files.
Setup.sh: Success!
BuildLibCarla.sh: Building LibCarla "Server.Release" configuration.
ninja: no work to do.
[0/1] Install the project...
-- Install configuration: "Server"
BuildLibCarla.sh: Success!
BuildCarlaUE4.sh: Using Unreal Engine at '/home/abdallahaymaan/UnrealEngine-4.24'
BuildCarlaUE4.sh: Build CarlaUE4 project.
make[1]: Entering directory '/home/abdallahaymaan/carla/Unreal/CarlaUE4'
bash "/home/abdallahaymaan/UnrealEngine-4.24/Engine/Build/BatchFiles/Linux/RunMono.sh" "/home/abdallahaymaan/UnrealEngine-4.24/Engine/Binaries/DotNET/UnrealBuildTool.exe" CarlaUE4Editor Linux Development  -project="/home/abdallahaymaan/carla/Unreal/CarlaUE4/CarlaUE4.uproject" 

Running Mono...

Fixing inconsistent case in filenames.
Setting up Mono
~/UnrealEngine-4.24/Engine ~/carla/Unreal/CarlaUE4
Using 'git status' to determine working set for adaptive non-unity build (/home/abdallahaymaan/UnrealEngine-4.24).
Creating makefile for CarlaUE4Editor (no existing makefile)
Parsing headers for CarlaUE4Editor
  Running UnrealHeaderTool "/home/abdallahaymaan/carla/Unreal/CarlaUE4/CarlaUE4.uproject" "/home/abdallahaymaan/carla/Unreal/CarlaUE4/Intermediate/Build/Linux/B4D820EA/CarlaUE4Editor/Development/CarlaUE4Editor.uhtmanifest" -LogCmds="loginit warning, logexit warning, logdatabase error" -Unattended -WarningsAsErrors
/home/abdallahaymaan/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Actor/ActorAttribute.h(29) : LogCompile: Error: 'EActorAttributeType' does not have a 0 entry! (This is a problem when the enum is initalized by default)
/home/abdallahaymaan/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Vehicle/VehicleInputPriority.h(24) : LogCompile: Error: 'EVehicleInputPriority' does not have a 0 entry! (This is a problem when the enum is initalized by default)
/home/abdallahaymaan/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Util/ActorAttacher.h(27) : LogCompile: Error: 'EAttachmentType' does not have a 0 entry! (This is a problem when the enum is initalized by default)
/home/abdallahaymaan/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Settings/QualityLevelUE.h(27) : LogCompile: Error: 'EQualityLevel' does not have a 0 entry! (This is a problem when the enum is initalized by default)
/home/abdallahaymaan/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Traffic/TrafficSignBase.h(31) : LogCompile: Error: 'ETrafficSignState' does not have a 0 entry! (This is a problem when the enum is initalized by default)
Makefile:290: recipe for target 'CarlaUE4Editor' failed
make[1]: *** [CarlaUE4Editor] Error 5
make[1]: Leaving directory '/home/abdallahaymaan/carla/Unreal/CarlaUE4'
Util/BuildTools/Linux.mk:7: recipe for target 'launch' failed
make: *** [launch] Error 2

any possible solutions please?

amstrudy, 2020-03-04:
What OS are you running? What version of CARLA? Have you followed the instructions [here](https://carla.readthedocs.io/en/latest/build_linux/) for building on Linux? Have you tried re-cloning and rebuilding?

hortonwu, 2020-03-24:
I run to the same issue. re-cloning and rebuilding doesn't work

Jack-Pinnington, 2020-04-03:
I've got the same error - make launch returns the "EActorAttributeType does not have a 0 entry!" error. Any help would be appreciated!

luroess, 2020-04-08:
Reinstalling all packages solved the problem for me.
Ubuntu Bionic:
`sudo apt install --reinstall build-essential clang-8 lld-8 g++-7 cmake ninja-build libvulkan1 python python-pip python-dev python3-dev python3-pip libpng-dev libtiff5-dev libjpeg-dev tzdata sed curl unzip autoconf libtool rsync libxml2-dev`

Jack-Pinnington, 2020-04-08:
Just tried reinstalling all packages using the command above - still no success. Called "make launch" and "make rebuild", to no success. Any help would be appreciated! Using CARLA 0.9.8 and following these instructions:

https://carla.readthedocs.io/en/latest/build_linux/

Jack-Pinnington, 2020-04-16:
I've managed to successfully complete "make launch" - new version of the source code has been released which clearly uses a 0 enum:

   17 /// List of valid types for actor attributes.
   18 UENUM(BlueprintType)
   19 enum class EActorAttributeType : uint8
   20 {
   21   Null = 0, // Workarround for UE4.24 issue with enums
   22   Bool      = CARLA_ENUM_FROM_RPC(Bool)      UMETA(DisplayName = "Bool"),
   23   Int       = CARLA_ENUM_FROM_RPC(Int)       UMETA(DisplayName = "Integer"),
   24   Float     = CARLA_ENUM_FROM_RPC(Float)     UMETA(DisplayName = "Float"),
   25   String    = CARLA_ENUM_FROM_RPC(String)    UMETA(DisplayName = "String"),
   26   RGBColor  = CARLA_ENUM_FROM_RPC(RGBColor)  UMETA(DisplayName = "RGB Color (comma separated)"),
   27 
   28   SIZE        UMETA(Hidden),
   29   INVALID     UMETA(Hidden)
   30 };

Seems like the only fix is to redo the Linux build from the "git clone <carla>" step. Hope this has helped.

ludennis, 2020-04-21:
> I've managed to successfully complete "make launch" - new version of the source code has been released which clearly uses a 0 enum:
> 
> 17 /// List of valid types for actor attributes.
> 18 UENUM(BlueprintType)
> 19 enum class EActorAttributeType : uint8
> 20 {
> 21 Null = 0, // Workarround for UE4.24 issue with enums
> 22 Bool = CARLA_ENUM_FROM_RPC(Bool) UMETA(DisplayName = "Bool"),
> 23 Int = CARLA_ENUM_FROM_RPC(Int) UMETA(DisplayName = "Integer"),
> 24 Float = CARLA_ENUM_FROM_RPC(Float) UMETA(DisplayName = "Float"),
> 25 String = CARLA_ENUM_FROM_RPC(String) UMETA(DisplayName = "String"),
> 26 RGBColor = CARLA_ENUM_FROM_RPC(RGBColor) UMETA(DisplayName = "RGB Color (comma separated)"),
> 27
> 28 SIZE UMETA(Hidden),
> 29 INVALID UMETA(Hidden)
> 30 };
> 
> Seems like the only fix is to redo the Linux build from the "git clone " step. Hope this has helped.

For those who need it, the commit is: "4384e386c21d037a951324b9885fe6a457e69e38"

germanros1987, 2020-06-26:
Is this still a problem?

bernatx, 2020-07-01:
Hi,
I remember that problem when trying to run CARLA 0.9.8 with UE 4.24
Remember that CARLA 0.9.8 runs with UE 4.23, and CARLA 0.9.9 runs with UE 4.24

Can you tell if that is the case? Are you running CARLA 0.9.8 with UE 4.24?

Regards

germanros1987, 2020-07-30:
Issue closed due to lack of activity. Feel free to open a new issue if you still consider this is a critical matter.

closed 2020-07-30 by germanros1987

qhaas, 2020-09-08:
> Remember that CARLA 0.9.8 runs with UE 4.23, and CARLA 0.9.9 runs with UE 4.24

IIRC, CARLA 0.9.8 uses UE 4.22+clang-7 and CARLA 0.9.9.x uses UE 4.24+clang-8.  You can get CARLA 0.9.8 working with UE 4.23.1 (and clang-8) with some minor changes.  However, CARLA 0.9.9.x uses UE 4.24 features that don't exist in versions prior to it.

filippogandolfi, 2020-10-06:
> Hi,
> I remember that problem when trying to run CARLA 0.9.8 with UE 4.24
> Remember that CARLA 0.9.8 runs with UE 4.23, and CARLA 0.9.9 runs with UE 4.24
> 
> Can you tell if that is the case? Are you running CARLA 0.9.8 with UE 4.24?
> 
> Regards

Hi @bernatx, I make a mistake and the UE4 project push me out with an error. I went back to previous version nbut now I have this problem 
`[29/34] Module.Carla.cpp
  ...Carla.cpp(87): error C4067: unexpected tokens following preprocessor directive - expected a newline
  .../PrepareAssetsForCookingCommandlet.cpp(55): warning C4458: declaration of 'TerrainNodeMaterial' hides class member
  ...
  .../DVSCamera.cpp(143): warning C4458: declaration of 'Stream' hides class member
  ...
  .../Response.h(50): note: while trying to match the argument list '(_Ty)'
          with
          [
              _Ty=bool
          ]
  .../CarlaServer.cpp(163): note: see reference to function template instantiation 'carla::rpc::Response<bool>::Response<bool>(TValue &&)' being compiled
          with
          [
              TValue=bool
          ]
  .../CarlaServer.cpp(163): note: see reference to function template instantiation 'carla::rpc::Response<bool>::Response<bool>(TValue &&)' being compiled
          with
          [
              TValue=bool
          ]
  .../CompositeSerializer.h(29): note: see reference to class template instantiation 'carla::sensor::CompileTimeTypeMap<std::pair<ACollisionSensor *,carla::sensor::s11n::CollisionEventSerializer>,std::pair<ADepthCamera *,carla::sensor::s11n::ImageSerializer>,std::pair<ADVSCamera *,carla::sensor::s11n::DVSEventArraySerializer>,std::pair<AGnssSensor *,carla::sensor::s11n::GnssSerializer>,std::pair<AInertialMeasurementUnit *,carla::sensor::s11n::IMUSerializer>,std::pair<ALaneInvasionSensor *,carla::sensor::s11n::NoopSerializer>,std::pair<AObstacleDetectionSensor *,carla::sensor::s11n::ObstacleDetectionEventSerializer>,std::pair<ARadar *,carla::sensor::s11n::RadarSerializer>,std::pair<ARayCastLidar *,carla::sensor::s11n::LidarSerializer>,std::pair<ARssSensor *,carla::sensor::s11n::NoopSerializer>,std::pair<ASceneCaptureCamera *,carla::sensor::s11n::ImageSerializer>,std::pair<ASemanticSegmentationCamera *,carla::sensor::s11n::ImageSerializer>,std::pair<FWorldObserver *,carla::sensor::s11n::EpisodeStateSerializer>>' being compiled
  D:/Tesisti/Filippo_Gandolfi/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Sensor/SensorFactory.cpp(44): note: see reference to class template instantiation 'carla::sensor::CompositeSerializer<std::pair<ACollisionSensor *,carla::sensor::s11n::CollisionEventSerializer>,std::pair<ADepthCamera *,carla::sensor::s11n::ImageSerializer>,std::pair<ADVSCamera *,carla::sensor::s11n::DVSEventArraySerializer>,std::pair<AGnssSensor *,carla::sensor::s11n::GnssSerializer>,std::pair<AInertialMeasurementUnit *,carla::sensor::s11n::IMUSerializer>,std::pair<ALaneInvasionSensor *,carla::sensor::s11n::NoopSerializer>,std::pair<AObstacleDetectionSensor *,carla::sensor::s11n::ObstacleDetectionEventSerializer>,std::pair<ARadar *,carla::sensor::s11n::RadarSerializer>,std::pair<ARayCastLidar *,carla::sensor::s11n::LidarSerializer>,std::pair<ARssSensor *,carla::sensor::s11n::NoopSerializer>,std::pair<ASceneCaptureCamera *,carla::sensor::s11n::ImageSerializer>,std::pair<ASemanticSegmentationCamera *,carla::sensor::s11n::ImageSerializer>,std::pair<FWorldObserver *,carla::sensor::s11n::EpisodeStateSerializer>>' being compiled
...  
.../Response.h(50): note: while trying to match the argument list '(_Ty)'
          with
          [
              _Ty=uint64
          ]
  .../CarlaServer.cpp(211): note: see reference to function template instantiation 'carla::rpc::Response<uint64_t>::Response<uint64>(TValue &&)' being compiled
          with
          [
              TValue=uint64
          ]
  .../CarlaServer.cpp(211): note: see reference to function template instantiation 'carla::rpc::Response<uint64_t>::Response<uint64>(TValue &&)' being compiled
          with
          [
              TValue=uint64
          ]
  .../Response.h(50): error C2665: 'boost::variant<carla::rpc::Response<carla::rpc::ActorId>::error_type,unsigned int>::variant': none of the 3 overloads could convert all the argument types
  .../variant.hpp(1767): note: could be 'boost::variant<carla::rpc::Response<carla::rpc::ActorId>::error_type,unsigned int>::variant(boost::variant<carla::rpc::Response<carla::rpc::ActorId>::error_type,unsigned int> &&) noexcept'
...
 .../apply_visitor_unary.hpp(99): note: see reference to function template instantiation 'CR carla::Functional::Recursive<FuncT>::operator ()<T&>(T &) const' being compiled
          with
          [
              FuncT=carla::Functional::Overload<FCarlaServer::FPimpl::BindActions::<lambda_0846187e3f40d40ff7159d5181d95228>,FCarlaServer::FPimpl::BindActions::
...
          ]
 ...
              Visitor=const carla::Functional::Recursive<carla::Functional::Overload<FCarlaServer::FPimpl::BindActions::<lambda_0846187e3f40d40ff7159d5181d95228>,FCarlaServer::FPimpl::BindActions::<lambda_ab41a6888e05bb87a95abe21bcc2857f>,FCarlaServer::FPimpl::BindActions::<lambda_97f57e95692603b937709ff77a9ca100>,FCarlaServer::FPimpl::BindActions::<lambda_370b3500c37927d6c917a7e7d3d41849>,FCarlaServer::FPimpl::BindActions::<lambda_4505cb02eea3d5f34ad45961a0a56d8f>,FCarlaServer::FPimpl::BindActions::<lambda_fc7487a4a460e3c3bf2ea559b671a5e1>,FCarlaServer::FPimpl::BindActions::<lambda_9e1493c50b42a3dbd68d435464f9d850 [... cut ...] tions::<lambda_f14ae1bbe0a4392ef38f022d72512610>,FCarlaServer::FPimpl::BindActions::<lambda_a93a273d5eb51aa3e8bec7a3d9ec60b2>,FCarlaServer::FPimpl::BindActions::<lambda_1edd184f520370ec0331481354132ce1>,FCarlaServer::FPimpl::BindActions::<lambda_f1fde5210522f4e0cfc5060bdb457710>,FCarlaServer::FPimpl::BindActions::<lambda_87a8f67edf14518153ce3db5cb7bca19>>> &,
              Visitable=const carla::rpc::Command::CommandType &
          ]
make: *** [CarlaUE4Editor] Error 5`
I'm on 0.9.9 and UE4.24.3

frankeng, 2020-10-07:
You can get CARLA 0.9.8 working with UE 4.23.1 (and clang-8) with some minor changes. 

Can somebody tell me how to do that, I want to use it with an other plug-in dat was made for 4.23

qhaas, 2020-10-07:
> Can somebody tell me how to do that, I want to use it with an other plug-in that was made for 4.23

If your plugin works with UE 4.24, then you can simply use newer versions of CARLA (0.9.9+) that were built against UE 4.24.

(disclaimer:  using a different version of UnrealEngine / Clang than the one officially supported by the target version of CARLA can lead to unexpected behavior)

Now, to answer your question...

You will have to compile CARLA from source and use CARLA 0.9.8 or older, since CARLA 0.9.9+ leverages new UE 4.24 API calls.  To get CARLA 0.9.8 working with clang-8, [apply these changes](https://github.com/carla-simulator/carla/commit/457b63b85ef9227f36f7a3dee07da34c1c8d9276).  To get it working with UE 4.23.1, you will just need to comment out about 4 lines of unneeded code in the sensor code the compiler will point you to.

However, UE4.23 doesn't like some of the assets of CARLA 0.9.8 (likely because they were made with UE 4.22 in mind), so you will either have to run it via the UnrealEditor or fix whatever is causing these errors to occur with `make package`:

```
LogInit: Display: LogBlueprint: Error: [Compiler BaseVehiclePawn] The current value of the ' Target Array ' pin is invalid: Array inputs (like 'Target Array') must have an input wired into them (try connecting a MakeArray node). from Source: /Game/Carla/Blueprints/Vehicles/BaseVehiclePawn.BaseVehiclePawn
LogInit: Display: LogBlueprint: Error: [Compiler BaseVehiclePawn] The current value of the ' Target Array ' pin is invalid: Array inputs (like 'Target Array') must have an input wired into them (try connecting a MakeArray node). from Source: /Game/Carla/Blueprints/Vehicles/BaseVehiclePawn.BaseVehiclePawn
LogInit: Display: LogBlueprint: Error: [Compiler BaseVehiclePawn] The current value of the ' Array ' pin is invalid: Array inputs (like 'Array') must have an input wired into them (try connecting a MakeArray node). from Source: /Game/Carla/Blueprints/Vehicles/BaseVehiclePawn.BaseVehiclePawn
LogInit: Display: LogBlueprint: Error: [Compiler BaseVehiclePawn] The current value of the ' Array ' pin is invalid: Array inputs (like 'Array') must have an input wired into them (try connecting a MakeArray node). from Source: /Game/Carla/Blueprints/Vehicles/BaseVehiclePawn.BaseVehiclePawn
```

frankeng, 2020-10-07:
@qhaas Thank You, I will try

frankeng, 2020-10-08:
Unfortunately no success.
 I got the messsage to buil it manually.
I try to compile and run it under Windows 10, So I opene visual studio 2017 as an administrator and tried to build it:

1>------ Build started: Project: CarlaUE4, Configuration: Development_Editor x64 ------
2>------ Skipped Build: Project: UE4, Configuration: BuiltWithUnrealBuildTool Win32 ------
2>Project not selected to build for this solution configuration 
1>Creating makefile for CarlaUE4Editor (no existing makefile)
1>Parsing headers for CarlaUE4Editor
1>  Running UnrealHeaderTool "C:\carla\Unreal\CarlaUE4\CarlaUE4.uproject" "C:\carla\Unreal\CarlaUE4\Intermediate\Build\Win64\CarlaUE4Editor\Development\CarlaUE4Editor.uhtmanifest" -LogCmds="loginit warning, logexit warning, logdatabase error" -Unattended -WarningsAsErrors -installed
1>Reflection code generated for CarlaUE4Editor in 5.4677943 seconds
1>Using Visual Studio 2017 14.16.27023 toolchain (C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023) and Windows 10.0.17763.0 SDK (C:\Program Files (x86)\Windows Kits\10).
1>Building 22 actions with 12 processes...
1>  [1/22] Default.rc2
1>  [2/22] Default.rc2
1>  [3/22] Default.rc2
1>  [4/22] SharedPCH.UnrealEd.cpp
1>  [5/22] Module.Carla.gen.4_of_4.cpp
1>  [6/22] Module.CarlaExporter.cpp
1>  [7/22] UE4Editor-CarlaExporter.lib
1>     Creating library C:\carla\Unreal\CarlaUE4\Plugins\CarlaExporter\Intermediate\Build\Win64\UE4Editor\Development\CarlaExporter\UE4Editor-CarlaExporter.lib and object C:\carla\Unreal\CarlaUE4\Plugins\CarlaExporter\Intermediate\Build\Win64\UE4Editor\Development\CarlaExporter\UE4Editor-CarlaExporter.exp
1>  [8/22] UE4Editor-CarlaExporter.dll
1>     Creating library C:\carla\Unreal\CarlaUE4\Plugins\CarlaExporter\Intermediate\Build\Win64\UE4Editor\Development\CarlaExporter\UE4Editor-CarlaExporter.suppressed.lib and object C:\carla\Unreal\CarlaUE4\Plugins\CarlaExporter\Intermediate\Build\Win64\UE4Editor\Development\CarlaExporter\UE4Editor-CarlaExporter.suppressed.exp
1>  [9/22] Module.Carla.gen.3_of_4.cpp
1>  [10/22] PCH.CarlaUE4.cpp
1>  [11/22] Module.Carla.gen.1_of_4.cpp
1>  [12/22] CarlaUE4.cpp
1>  [13/22] CarlaUE4.init.gen.cpp
1>  [14/22] CarlaMapGenerator.gen.cpp
1>  [15/22] CarlaMapGenerator.cpp
1>  [16/22] UE4Editor-CarlaUE4.lib
1>     Creating library C:\carla\Unreal\CarlaUE4\Intermediate\Build\Win64\UE4Editor\Development\CarlaUE4\UE4Editor-CarlaUE4.lib and object C:\carla\Unreal\CarlaUE4\Intermediate\Build\Win64\UE4Editor\Development\CarlaUE4\UE4Editor-CarlaUE4.exp
1>  [17/22] Module.Carla.gen.2_of_4.cpp
1>  [18/22] Module.Carla.cpp
1>C:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Commandlet/PrepareAssetsForCookingCommandlet.cpp(33): warning C4458: declaration of 'TerrainNodeMaterial' hides class member
1>  c:\carla\unreal\carlaue4\plugins\carla\source\carla\commandlet\PrepareAssetsForCookingCommandlet.h(161): note: see declaration of 'UPrepareAssetsForCookingCommandlet::TerrainNodeMaterial'
1>C:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Recorder/CarlaRecorder.cpp(347): warning C4458: declaration of 'Role' hides class member
1>  C:\Program Files\Epic Games\UE_4.23\Engine\Source\Runtime\Engine\Classes\GameFramework/Actor.h(448): note: see declaration of 'AActor::Role'
1>C:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Recorder/CarlaRecorder.cpp(356): warning C4458: declaration of 'Role' hides class member
1>  C:\Program Files\Epic Games\UE_4.23\Engine\Source\Runtime\Engine\Classes\GameFramework/Actor.h(448): note: see declaration of 'AActor::Role'
1>C:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Recorder/CarlaRecorderQuery.cpp(45): warning C4996: 'localtime': This function or variable may be unsafe. Consider using localtime_s instead. To disable deprecation, use _CRT_SECURE_NO_WARNINGS. See online help for details.
1>  C:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\ucrt\time.h(506): note: see declaration of 'localtime'
1>C:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Sensor/CollisionSensor.cpp(49): warning C4458: declaration of 'Episode' hides class member
1>  C:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla/Sensor/Sensor.h(99): note: see declaration of 'ASensor::Episode'
1>C:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Sensor/GnssSensor.cpp(51): warning C4458: declaration of 'Stream' hides class member
1>  C:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla/Sensor/Sensor.h(97): note: see declaration of 'ASensor::Stream'
1>C:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Sensor/InertialMeasurementUnit.cpp(48): warning C4458: declaration of 'Owner' hides class member
1>  C:\Program Files\Epic Games\UE_4.23\Engine\Source\Runtime\Engine\Classes\GameFramework/Actor.h(431): note: see declaration of 'AActor::Owner'
1>C:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Sensor/InertialMeasurementUnit.cpp(178): warning C4458: declaration of 'Stream' hides class member
1>  C:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla/Sensor/Sensor.h(97): note: see declaration of 'ASensor::Stream'
1>C:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Sensor/ObstacleDetectionSensor.cpp(138): warning C4458: declaration of 'Episode' hides class member
1>  C:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla/Sensor/Sensor.h(99): note: see declaration of 'ASensor::Episode'
[... cut ...]
1>  C:\Program Files\Epic Games\UE_4.23\Engine\Source\Runtime\Engine\Classes\GameFramework/Controller.h(76): note: see declaration of 'AController::Character'
1>C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\VC\VCTargets\Microsoft.MakeFile.Targets(44,5): error MSB3075: The command ""C:\Program Files\Epic Games\UE_4.23\Engine\Build\BatchFiles\Build.bat" CarlaUE4Editor Win64 Development -**Project="C:\carla\Unreal\CarlaUE4\CarlaUE4.uproject" -WaitMutex -FromMsBuild" exited with code 5. Please verify that you have sufficient rights to run this command.
1>Done building project "CarlaUE4.vcxproj" -- FAILED.**

qhaas, 2020-10-08:
> Unfortunately no success.

I should have pointed out that `PostPhysicsComponentTick.bCanEverTick` occurs in more files than just the 'Sensor.cpp' I linked, such as [RssSensor.cpp](https://github.com/carla-simulator/carla/commit/6ba4d68a48cb6af2ac5d5983e6e44cd5ebb54777#diff-8169d3a8a9eb74e8aa3110d488bb578b), which is causing one of the errors in that trace.

parth3153, 2020-12-22:
sudo update-alternatives --install /usr/bin/clang++ clang++ /usr/lib/llvm-8/bin/clang++ 180 &&
sudo update-alternatives --install /usr/bin/clang clang /usr/lib/llvm-8/bin/clang 180

this worked for me, To avoid compatibility issues between Unreal Engine and the CARLA dependencies, use the same compiler version and C++ runtime library to compile everything. The CARLA team uses clang-8 and LLVM's libc++. Change the default clang version to compile Unreal Engine and the CARLA dependencies.

source: https://carla.readthedocs.io/en/latest/build_linux/#linux-build-command-summary

darkdevahm, 2021-09-13:
I'm currently running into the same issue when running "make launch"

8 warnings and 4 errors generated.
7 warnings and 6 errors generated.
Makefile:315: recipe for target 'CarlaUE4Editor' failed
make[1]: *** [CarlaUE4Editor] Error 6
make[1]: Leaving directory '/users/anasrahm/carla/Unreal/CarlaUE4'
Util/BuildTools/Linux.mk:7: recipe for target 'launch' failed
make: *** [launch] Error 2
anasrahm@node0:~/carla$ Util/BuildTools/Linux.mk:7: recipe for target 'launch' failed
-bash: Util/BuildTools/Linux.mk:7:: No such file or directory
anasrahm@node0:~/carla$ make: *** [launch] Error 2

Any advises?

zzningxp, 2022-07-12:
From: https://github.com/carla-simulator/carla/issues/4907

Just to clarify, the UE4 that you need to use with CARLA 0.9.13 is this one (UE4 fork):  https://github.com/CarlaUnreal/UnrealEngine

Do not use the official UnrealEngine version, you should use **UnrealEngine-Carla** version.

yuan-mingxi, 2024-02-28:
> sudo update-alternatives --install /usr/bin/clang++ clang++ /usr/lib/llvm-8/bin/clang++ 180 && sudo update-alternatives --install /usr/bin/clang clang /usr/lib/llvm-8/bin/clang 180
>
> this worked for [...]

omg,you save my day……the UE successfully open without crash