carla-simulator/carla #2559: How can I get a specific traffic light with python api?

Zhuxy, 2020-03-06:
I want to control traffic lights manually. There is ActorList.filter("traffic.traffic_light") to get all of them. But how can I get a specific one in the map? 

Maybe mark or add some tag to traffic lights in Unreal scene editor and filter by them in pyhton client API? 

Is that possible?

many thanks

hepeng187, 2020-04-28:
I encountered the same problem.  Have you solved the problem of setting only one traffic light?

germanros1987, 2020-06-26:
@Axel1092 could you provide some guidance for this?

Axel1092, 2020-07-03:
Hi @Zhuxy,
We plan to expose traffic lights through the landmark class which allows you to detect sign and traffic light in the OpenDRIVE file. This is currently working with custom OpenDRIVE files loaded through the standalone mode but it is not yet avaialiable in Carla's maps. You can still sort the traffic lights by position in the world `traffic_light.get_location()`.

closed 2020-09-11 by stale[bot]