carla-simulator/carla #2562: How to build map using Ubuntu 18.04?

EgorFollia, 2020-03-07:
Hi! 
I was wondering whether it's possible to create a map (or change the existing one) using UE4?
Should I use RoadRunner? I can't find any documentation on how to do it.

Maybe there are some maps I can import? Provide me, please, with the source
P.S. Thanks a lot in advance

stevelaclasse, 2020-03-09:
you could easily edit the existing one, for that you must build Carla from the source, and launch UE4editor to edit Map and then create a package with your new edited Maps.

EgorFollia, 2020-03-10:
@stevelaclasse I should run this in carla or in UE4 directory?

stevelaclasse, 2020-03-11:
in carla with the `make launch` or `make launch-only` command.

EgorFollia, 2020-03-11:
@stevelaclasse yes, that was a dummy one question, sorry )
I have install PythonAPI today and now after I `make launch` Carla - it just freezes my PC
What seems to be the problem?

closed 2020-03-13 by EgorFollia