carla-simulator/carla #2565: Inappropriate behaviour in high speed automatic driving

barasm-hita, 2020-03-09:
Hi everyone,
I have a problem at running **automatic_control.py**.
The first thing to say, I use CARLA version 0.9.7 but not the latest commit. I got the source zip file from release page. Also, have to say that I built it on Windows 10 via VS2019 (made some changes in **cmake** files to change VS2017 to VS2019 by myself) and running with unreal engine 4.22.

Now let me describe what is the problem.
I need to check autonomous driving vehicles behaviour on speeds more than 50 km/h, then I changed **target_speed to 50** in file **basic_agent.py** and get automatic_control.py running again. **_I found my vehicle driving like a drunk guy!!_**

[image]

Please If you know why it should happen, help me.
Thanks for any replies

jackbart94, 2020-03-09:
Hi,

Basic agent is pretty... basic. Probably needs a PID re-tuning, since you change its target velocity. Try the Roaming agent or the [Behavior agent](https://github.com/carla-simulator/carla/tree/jackbart94-agent-behaviors)

barasm-hita, 2020-03-10:
Thanks a lot @jackbart94 
But there is again the same problem while I made target_speed 50 km/h in local_planner.py and ran automatic_control.py with Roaming agent.

Also, can I get target_speed dynamically from speed limit signs?

jackbart94, 2020-03-10:
Hi,

You still need to change the PID values once you change the desired velocity.

Also yes, you can, just need to get the current affecting speed limit on your ego vehicle. If you don't know how, check the documentation!

barasm-hita, 2020-03-10:
@jackbart94 
Thanks again for the very fast response. I found some solutions in your documentation.
I'm going to close this issue.
Have a nice time.
Bye

closed 2020-03-10 by barasm-hita