carla-simulator/carla #2567: kaggle or colab support

khojir, 2020-03-09:
Hi.
Is there a way to run Carla on Kaggle or Google Colab? I think there is a lot of students that can't run Carla on their laptop. It would be nice if you support these options.
Thanks.

MichaelBosello, 2020-04-20:
I needed to run Carla on Colab too, and I have found a method to do so.
If you're still interested, I did a Colab notebook with the tutorial.
You can find it at https://github.com/MichaelBosello/carla-colab

khojir, 2020-05-29:
> I needed to run Carla on Colab too, and I have found a method to do so.
> If you're still interested, I did a Colab notebook with the tutorial.
> You can find it at https://github.com/MichaelBosello/carla-colab

I tried this solution and it works really good. Thank you so much.

closed 2020-05-29 by khojir

Sukhman723, 2020-05-30:
Hi khojir, it is showing unable to run nagrok

harish-2306, 2020-06-20:
> Hi khojir, it is showing unable to run nagrok 

pip install the library

wendersonj, 2020-08-27:
Thansk for sharing.
I'm getting a error while it's fetching a library from nvidia (fetch: libnvidia-decode-450). Then I started a issue on their github:
https://github.com/demotomohiro/remocolab/issues/59