carla-simulator/carla #2568: Python API build fails (CARLA 0.9.7) in Windows:

danielaDESM, 2020-03-09:
Hi!
I have been trying to build CARLA the last days. I have gone through the FAQS several times and I have encountered several issues in the way and tried some solutions that others have posted. Currently, I think the only problem left is: make Python API. 

I am using: 
- Windows 10
- Python 3.7.3
- x64 Native Tools for VS 2017
- Unreal Engine 4.22.3
- GNU make 3.81
- Cmake 3.14.0-rc4

After I run the command, it starts compiling until it starts apparently complaining about boost:

[image]

Then, for many .cpp files it shows me "class template optional is only available with C++17 or later."
[image]

And finally, it throws a lot of linking issues:

[image]

[image]

As a note side, I was not able to do make launch either. I was having the same message of "class template optional is only available with C++17 or later. ". But following other posts I have erased some "uint" and exchange them by "uint_64" in TrafficManagerRemote.cpp. I have erased several time times the whole project and this trial has been the farthest I can get. 

I hope you can help me,

barasm-hita, 2020-03-10:
Hi @danielaDESM 
This problem you are facing causes from Unreal Engine registry key issue (as you got error "Unreal Engine 4.22 not detected" in the last screenshot). So, it can be solved by adding the registry key named **InstalledDirectoy** (and type of _String Value_) in **Computer\HKEY_LOCAL_MACHINE\SOFTWARE\EpicGames\Unreal Engine\4.22** like how you see in my screenshot mentioned.
If any more details were required, tell me to explain.
Good luck.

[image]

danielaDESM, 2020-03-10:
Hi!
Thanks for your help. Just one more question. I do not have the folder "EpicGames/Unreal Engine/4.22" under HKEY_LOCAL_MACHINE/SOFTWARE. Instead, I have found it here:
[image]
Can I add the value there?

danielaDESM, 2020-03-10:
So I did change the value in "InstalledDirectory" under HKEY_LOCAL_MACHINE/SOFTWARE/WOW6432-node/EpicGames/Unreal Engine/4.0. The error that you mentioned, does not appear anymore. Nonetheless, previous errors are still there. 
I just ran make "make PythonAPI" and I have the  "class template optional is only available with C++17 or later" and the linking errors:
[image]

[image]

barasm-hita, 2020-03-10:
Hello again @danielaDESM 
The subject "class template optional is only available with C++17 or later" is not an error, it's just a warning and it doesn't make any problem in your work.

About LNK errors that you got, I'm not sure what has made them occur, but it's better to first check your **boost-install** folder in the **Build** directory. It shouldn't be empty!

microbirdman, 2020-03-10:
Hi,

I am having the exact same problem with Boost not being linked properly. I tried to compile 0.9.5, 0.9.6, 0.9.7 and I am always getting that error. In fact, I was trying to compile 0.9.8 just now and it didn't work either. Please see attached screenshot.

[image]

I tried different versions of Python and Cmake but still not working. Any ideas? Thanks in advance.

Julian9875, 2020-03-10:
I'm having the same issue, it should has been fixed according to the documentation, but it's not :(

danielaDESM, 2020-03-12:
I have download everything again and boost appears to be installed correctly. After make launch, unreal ask me to compile again some files and then it shows an error. I open the project in Visual Studio and after building it it only shows me this error. Do someone know how to tackle this?
[image]

(effectively if I look into the folder carla/Version.h does not exist)

danielaDESM, 2020-03-12:
I have copied Version.h from another folder (it has a generic code) and it runs without problems. It opens Unreal with no issues. I play it and everything looks fine. When I continue with the make Python API, it shows me an error of not finding Version.h under carla/, so I check in the folder and there is a file "Version.h.in", so I just erased the ".in" extension and tried again to execute make PythonAPI, which leads me to error I was having previous times: the linking error.

[image]

closed 2020-03-12 by danielaDESM

OnTheStormx, 2020-03-14:
Did u find the solution of make PythonAPI @danielaDESM

barasm-hita, 2020-03-14:
Hi @OnTheStormx 
If the error you are facing is about missing file 'Version.h', the solution is how she told us in her last comment. Let me briefly explain it to you.
There should be a file named **'Version.h.in'** in your CARLA directory. You have to just remove the **'.in'** from the end of its name (rename it to **'Version.h'**). Then your error must be fixed.

estebanmoreno, 2020-03-20:
Hi @barasm-hita,

That's right. The Version.h.in has to be renamed. I had already done that on my previous post but still getting all the dependency errors I showed. Did you find a solution @danielaDESM?

estebanmoreno, 2020-03-20:
Hi @barasm-hita,

Thanks for the prompt reply. Actually I ran the **make** command from the root CARLA folder. I was at that other directory in the screenshot because I was trying to manually build it using setup.py.