carla-simulator/carla #2569: Cannot Spawn vehicle with attribute 'role_name' == 'hero'

Jack-Pinnington, 2020-03-09:
Hello,

I'd like to generate a log file which contains the information about 100 vehicles. This is so I can repeat data collection from a variety of sensors across different weather conditions in the same environment to measure Localization Algorithm performance across different weather conditions.

In order to find my vehicle to attach sensors to, I'd like to spawn one vehicle with an attribute 'role_name' == 'hero' which can be easily identified in the actorList. Here's the code I'm using:

spawn_points = world.get_map().get_spawn_points()
batch = []
    #Spawn 1 ego vehicle (vehicle.audi.tt)
    hero_bp = blueprint_library.find('vehicle.audi.tt')
    hero_transform = spawn_points[0]
    hero_bp.set_attribute('role_name', 'hero')
    if hero_bp.has_attribute('driver_id'):
        driver_id = random.choice(blueprint.get_attribute('driver_id').recommended_values)
        blueprint.set_attribute('driver_id', driver_id)
    batch.append(SpawnActor(hero_bp, hero_transform).then(SetAutopilot(FutureActor, True)))
    client.apply_batch_sync(batch)
    time.sleep(1)
    
    #Check I have created an egoVehicle...
    egoFlag = "false"
    actorList = world.get_actors()
    for actor in actorList:
	if(actor.attributes.get('role_name' == 'hero')):
		print("egoVehicle found!\n")
		egoFlag = "true"
		break

    if(egoFlag == "false"):
	print("Could not find egoVehicle even though I just spawned it!")

This test keeps failing with message "Could not find egoVehicle even though I just spawned it!" - do you have any insight into why?

My full source code is linked below:
https://github.com/Jack-Pinnington/Carla_Project/blob/master/generateFreeDrivingLog.py

Best regards,
Jack

stevelaclasse, 2020-03-11:
`hero_bp.set_attribute('role_name', 'hero')`  here could be the problem, because hero_bp is a list (that may contain just one element), you need to use `hero_bp[0].set_attribute('role_name', 'hero')`  which is your car to set the role_name.

Jack-Pinnington, 2020-04-03:
Issue found - sorry for the delay in response, has been fixed for a while.... It's the way I'm finding the egoVehicle that's causing the error - "if(actor.attributes.get('role_name' == 'hero')):" should read "if(actor.attributes["role_name"] == 'hero'):" since the attributes returns a dictionary. Think there was a code change between versions - just moved from 0.9.6 to 0.9.8.

closed 2020-04-03 by Jack-Pinnington