carla-simulator/carla #2572: Depth Camera 's depth value jumps

minrui-hust, 2020-03-10:
I am using depth camera, but the depth value of the depth camera seems split into two parts.
And there is a gap between them, as denoted by the blue line. The gap move with the ego car.
Any one know the reason?
[image]

minrui-hust, 2020-03-11:
I found the reason! I am debug carla from vs code, some how, the asset corresponding to weather is not load correctly, which impose some strips on the view. I think the issue #2460 may be caused by the same reason.

closed 2020-03-11 by minrui-hust

animareversed, 2020-03-11:
Hi,
How do you visualize the image? Which values does it have?

minrui-hust, 2020-03-12:
I just use the PythonAPI example/manual_control.py, use 'N' to switch depth camera(default RGB).

animareversed, 2020-03-12:
So, if I am not wrong, your image is in logarithmic mode, right?

minrui-hust, 2020-03-12:
Yes, in logarithmic mode.

animareversed, 2020-03-12:
Ok thanks

TzabarDolev, 2023-09-17:
There is an small error in the formula which is given in the 'readthedocs' of the depth sensor.
I wrote about it the CarlaSimBlog - https://carlasimblog.wordpress.com/2023/09/16/sensors/