carla-simulator/carla #2574: vehicle.set_light_state() only allows one type of light state to be set at a time

whitesea1, 2020-03-10:
First of all thanks for tackling this issue! It is exciting to have access to the vehicle lights with the release of 0.9.8!

Playing around with it, I noticed that I can't set multiple lights on at the same time, with the exception of All. For example, I cannot have the low beams on and also turn on a turn signal, it is strictly one or the other. Without the ability to pick and choose, it is unlikely we'll be able to use the vehicle lights in a meaningful way, as it is pretty rare that only one light is needed at a time throughout an entire scenario. 

System:
Ubuntu 18.04
Python 3.5
Carla 0.9.8 (release version)
Unreal 4.22

Beskhue, 2020-03-12:
You can set multiple lights at the same time like this:

```python
vehicle.set_light_state(carla.VehicleLightState(carla.VehicleLightState.RightBlinker | carla.VehicleLightState.LowBeam))
```

Ideally, from Python 3.6 onwards, the enum would become a flag so you would be able to write:

```python
vehicle.set_light_state(carla.VehicleLightState.RightBlinker | carla.VehicleLightState.LowBeam)
```

whitesea1, 2020-03-12:
Thank you for that answer! This wasn't clear from the documentation.

closed 2020-03-12 by whitesea1