carla-simulator/carla #2575: wheel_config.ini file not found

tmofle, 2020-03-11:
Hello,

I am trying to use a Logitech G29 Steering Wheel. I have found several threats referencing the wheel_config.ini file that should be present in the carla folder. However, I can not find this file. Does this file download when CARLA is installed? Thank you, in advance for any assistance.

ujjawal-kumar, 2020-03-11:
create a file with name "wheel_config.ini" and paste these line:-

[G29 Racing Wheel]
steering_wheel = 0
clutch = 1
throttle = 2
brake = 3
handbrake = 4
reverse = 5

tmofle, 2020-03-12:
Thank you for your reply.  However, I am still getting a few error codes. (see error message below)
Any further assistance in this matter would be greatly appreciated.
  
Traceback (most recent call last):
   line 865, in <module>
    main()
 line 857, in main
    game_loop(args)
  line 787, in game_loop
    controller = DualControl(world, args.autopilot)
  line 232, in __init__
    self._parser.get('G29 Racing Wheel', 'steering_wheel'))
 line 780, in get
    d = self._unify_values(section, vars)
 line 1146, in _unify_values
    raise NoSectionError(section) from None
configparser.NoSectionError: No section: 'G29 Racing Wheel'

ujjawal-kumar, 2020-03-12:
> Thank you for your reply. However, I am still getting a few error codes. (see error message below)
> Any further assistance in this matter would be greatly appreciated.
> 
> Traceback (most recent call last):
> line 865, in
> main()
> line 857, in main
> game_loop(args)
> line 787, in game_loop
> controller = DualControl(world, args.autopilot)
> line 232, in **init**
> self._parser.get('G29 Racing Wheel', 'steering_wheel'))
> line 780, in get
> d = self._unify_values(section, vars)
> line 1146, in _unify_values
> raise NoSectionError(section) from None
> configparser.NoSectionError: No section: 'G29 Racing Wheel'

I think you maybe you miss "[G29 Racing Wheel]" this line in that file.

tmofle, 2020-03-12:
[steeringwheelcode.txt](https://github.com/carla-simulator/carla/files/4324381/steeringwheelcode.txt)
Here is a text file with the code I am using.

tmofle, 2020-03-12:
The line "G29 Racing Wheel" is in the file. However, after looking at the file again I believe I may be missing another file the code is calling. If this assumption is correct what information should the file "G29 Racing Wheel" contain? Should the drive for the steering wheel be located in the CARLA folder?

Thank you for all of your assistance.

ujjawal-kumar, 2020-03-13:
> The line "G29 Racing Wheel" is in the file. However, after looking at the file again I believe I may be missing another file the code is calling. If this assumption is correct what information [...]

add 'wheel_config.ini' in the parser read like this in your script file:- 
self._parser.read('/Users/slims/Documents/Python/CARLA/PythonAPI/examples/wheel_config.ini')

closed 2020-05-19 by stale[bot]