carla-simulator/carla #2577: Retrieve all active ActorBlueprints with PythonAPI (including static objects)

manolotis, 2020-03-11:
I would like to retrieve as many objects as I can from the simulation, specifically their coordinates and area they occupy (2D is enough). Other than the actors I spawn myself, when using World.get_actors() I can only retrieve the spectator and several other 'traffic.*' actors. However, in the existing simulation there seem to be many static objects as specified in the blueprints (e.g. 'static.prop.plantpotXX' or 'static.prop.benchXX'), which I cannot retrieve.

Is there any way to access all these objects with the Python API? I would be very interested in this information specially for buildings and streetlamps, but any static object in the scene would also help me.

Currently using the latest build at the time of writing: 0.9.8

jackbart94, 2020-03-11:
Hi,

Can [this link](https://carla.readthedocs.io/en/latest/bp_library/#static) solve your issue?

manolotis, 2020-03-11:
> Hi,
> 
> Can [this link](https://carla.readthedocs.io/en/latest/bp_library/#static) solve your issue?

Hi @jackbart94,
No, I am familiar with that link and it cannot solve my issue. 
I would like to retrieve coordinates of all (or as many as possible) static objects (e.g. buildings, streetlamps, trees...etc) that are in the scene _when the simulation starts_ (so not the blueprints I can spawn myself, the objects that are already there). 
Please let me know if I can clarify my question further.

manolotis, 2020-03-12:
It seems this cannot be done from the client side as of now.

closed 2020-03-12 by manolotis

hubing1994, 2020-04-28:
I have the same questions with you!  World.get_actors() only can get what you spawn by python scrips. Howerver, I want to add some cars in UE4 editor, then get there locations by the simulator