carla-simulator/carla #2584: Hiding Actors to Client

manipadmeom, 2020-03-12:
Hi,

I wish to limit the number of actors the server sends to the client for a given condition (for example, limit the result of client.get_world().get_actors() to the ones within a radius r from the hero vehicle). Can someone say what is the best to do it? Probably, in case this is too complicated, is there a way to hide the location/speed/etc information? For example, if I called client.get_world.get_actor(id).get_location() the result the could be (0,0,0) even the actor is valid and has a current location.

germanros1987, 2020-06-26:
Could you explain what you are *really* trying to do?

germanros1987, 2020-07-07:
Closing issue due to inactivity. Please, feel free to comment back if you need the issue reopened.

closed 2020-07-07 by germanros1987