carla-simulator/carla #2585: Problem with PythonAPI while building CARLA 0.9.8

Julian9875, 2020-03-12:
Hello, I've been trying to build CARLA. But I have encountered many problems. Right now, I have a problem with make Python API. I've been trying to build for every version since 0.9.5. Now with this 0.9.8 version this problem it is supposed to be solved, but I still have this problem:

[image]

I am using:
Windows 10
Python 3.7.3
x64 Native Tools for VS 2017
Unreal Engine 4.22.3
GNU make 3.81
Cmake 3.14.0-rc4

OnTheStormx, 2020-03-14:
Hi @Julian9875 Did u find the solution.

Julian9875, 2020-03-14:
Unfortunately no, I installed Ubuntu to make it work, but I still need it in Windows :(

carfloresp, 2020-04-04:
Hi @Julian9875, I faced the same problem and managed to get around it. After compiling Carla client and server, instead of "make PythonAPI", I did:

cd PythonAPI/carla
python setup.py install

This generates /dist folder and the ".egg" file within. I am able to run the .py examples now.
Hope this helps.

closed 2020-06-10 by stale[bot]