carla-simulator/carla #2587: Pedestrians are sinking down to half of their height in Carla Town03 and custom maps

chandeep11, 2020-03-13:
I've been testing the pedestrian spawning feature using manual control in Carla 0.9.7. In some areas of Town03, I can see that the pedestrians are not resting on the ground/road and instead they are sinking down to half of their height. Seems like a bug in this town.

Pedestrian spawning resting properly.
[image]

Pedestrian sinking down.
[image]

Even in the custom map I've created using road runner,I'm facing the same issue. Instead of the pedestrian resting properly/sinking to half (as it happens in Carla towns), here they are not resting on the ground and they are falling down through roads indefinitely. 
[image]

I'm not using the landscapes in my custom map and only the meshes generated using road runner. Cars are spawning properly, colliding with the roads and working fine. The issue is only with the pedestrians. 
Can this issue be fixed by using the landscape along with my custom map? Temporarily I'm fixing this issue with invisible blocking volumes under the road (which is not as proper as the Carla towns). Please help me out with where I'm missing in my custom map.

Zhuxy, 2020-04-10:
Same problem. 
Is that fixed?

chandeep11, 2020-04-22:
@Zhuxy No, not yet found the solution/cause. @TheNihilisticRobot could you help on this?

ptigas, 2020-06-13:
@chandeep11 "Temporarily I'm fixing this issue with invisible blocking volumes under the road (which is not as proper as the Carla towns)." - is this something you don on roadrunner or you're hacking Carla for that?

chandeep11, 2020-06-13:
@ptigas Those blocking volumes are added via the UE4 Editor(It's a UE4 feature) in the source version Carla and then made a binary out of it. I don't think you can do the same in compiled version directly.

germanros1987, 2020-06-26:
@XGodina this needs urgent attention.

avishek-mondal, 2020-06-26:
This could be related to the issue I noticed here https://github.com/carla-simulator/carla/issues/2976

XGodina, 2020-06-30:
@avishek-mondal @Zhuxy Hi, Sorry to answer late. I m working with that. There are some ways to check it, collisions, or the Spaw points where pedestrians are born. As soon as I can, I'll let you know.

XGodina, 2020-07-09:
@Zhuxy @chandeep11. I advise you to use the current version of CARLA0.9.9. Does this only happen in Town03?

chandeep11, 2020-07-14:
@XGodina Yes, This happens only in some parts of Town03 and the custom maps created with RoadRunner. I've a doubt, will the FBX from RoadRunner is used directly or some alterations do take place (like adding collision to road or landscape or any blocking volumes beneath the road) in Carla.?
P.S: I'm importing the RoadRunner model as it is without any modifications in UE4 and then testing the Pedestrian spawning feature which makes the pedestrians to sink down.

closed 2020-09-20 by stale[bot]

firas-elhag, 2024-05-23:
Hi @XGodina has the issue related to importing fbx file of roadrunner into Carla has been identified as the issue to sinking pedestrians as I am facing the same issue is related to Carla or roadrunner?