carla-simulator/carla #2588: Pedestrian crossing issue

ndeshp2s, 2020-03-13:
Hello. I am trying to simulate a scenario where pedestrians would cross roads. And with carla-0.9.7 I am able to do so. However, there are some instances (depending on the goal I guess) where a pedestrian instead of crossing the road keeps walking on the road (instead of side walk) to reach its goal. I tried making changes in the Navigation.cpp file by increasing the "AREA_ROAD_COST" variable value, but this seem to have no effect on the pedestrian behavior. Is there any other way by which I can enforce the pedestrian to walk on road only when required (for example, in case of crossing)

KasraMokhtari2112, 2020-04-11:
I just started working with CARLA. I have two basic questions:
1. Where can we find the related cpp codes? Are there in CARLA folders?
2. After modifying cpp codes, how can we incorporate this into CARLA?

Thank you so much!

> Hello. I am trying to simulate a scenario where pedestrians would cross roads. And with carla-0.9.7 I am able to do so. However, there are some instances (depending on the goal I guess) where a [...]

closed 2020-06-17 by stale[bot]

sbv198, 2022-06-13:
@ndeshp2s Could you please share your code or help me since i am working with similar scenarios