carla-simulator/carla #2597: Can we use two clients to listen to the same sensor

wx9698, 2020-03-14:
Can we use two clients to get the same sensor (etc. camera) to listen and fetch the data simultaneously? Will you provide this function in the future release version?

wx9698, 2020-03-20:
And also, I see the changes to FDataMultiStream in this PR https://github.com/carla-simulator/carla/pull/915. In the section of "Other C++ improvements". Why still keeping FDataStream there?

sergi-e, 2020-03-20:
I am pinning @nsubiron, because maybe he knows something about this specific issue.

nsubiron, 2020-03-23:
Streams are a bit faster than multi-streams. Streams are lock-free while multi-streams get a lock and iterate a list of clients. This part was carefully optimized because, for cameras, this is executed in the render thread (usually the bottleneck), thus any tiny gain here contributes to overall performance.

It's a tiny performance improvement only measurable in the streaming benchmarks, I imagine not at all noticeable in the common use cases, but that's why at the time I didn't change all the sensors to multi-streams.

wx9698, 2020-03-23:
I see, thank for the explanation. So I guess you are still going to keep it there without any change?

nsubiron, 2020-03-24:
Try raising the issue with the core team if you think is better to use multistreams (I'm no longer working for Carla). I would do some benchmarks too, low resolution images, lidar in no-rendering mode, those are some use cases that could get worse performance if any. Also check if this affects frame drops, that's a real issue in sync mode as currently may cause dead-locks.

closed 2020-03-24 by wx9698