carla-simulator/carla #2598: : The repository 'http://dist.carla.org/carla-0.9.7 bionic Release' does not have a Release file.

jediofgever, 2020-03-14:
: The repository 'http://dist.carla.org/carla-0.9.7 bionic Release' does not have a Release file.

Linux deb file installation is not successul . I dont know why I am getting this error; 

`: The repository 'http://dist.carla.org/carla-0.9.7 bionic Release' does not have a Release file.`

himanshugoswami, 2020-03-14:
We have updated the APT repository. Now you can install carla-0.9.7 using

sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys DB53A429E64554FC &&
sudo add-apt-repository "deb [trusted=yes] http://dist.carla.org/carla-0.9.7/ all main"

Or Carla-0.9.8 using 

sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 304F9BC29914A77D &&
sudo add-apt-repository "deb [trusted=yes] http://dist.carla.org/carla-0.9.8/ all main"

egreglee, 2020-03-15:
There might still be a problem:

sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys DB53A429E64554FC && sudo add-apt-repository "deb [trusted=yes] http://dist.carla.org/carla-0.9.7/ all main"
[sudo] password for user: 
Sorry, try again.
[sudo] password for user: 
Executing: /tmp/apt-key-gpghome.eB8vLC9xNi/gpg.1.sh --keyserver keyserver.ubuntu.com --recv-keys DB53A429E64554FC
gpg: key 35615F47EA989526: "Carla Simulator <carla.simulator@gmail.com>" not changed
gpg: Total number processed: 1
gpg:              unchanged: 1
Ign:1 http://dl.google.com/linux/chrome/deb stable InRelease
Hit:2 http://us.archive.ubuntu.com/ubuntu bionic InRelease                                                                      
Hit:3 http://dl.google.com/linux/chrome/deb stable Release                                                                      
Ign:4 http://dist.carla.org/carla-0.9.7 bionic InRelease                                                            
Get:5 http://us.archive.ubuntu.com/ubuntu bionic-updates InRelease [88.7 kB]                  
Hit:6 http://dist.carla.org/carla-0.9.7 all InRelease                                                    
Get:7 http://security.ubuntu.com/ubuntu bionic-security InRelease [88.7 kB]                             
Err:8 http://dist.carla.org/carla-0.9.7 bionic Release                                                              
  404  Not Found [IP: 34.227.255.250 80]
Get:10 http://us.archive.ubuntu.com/ubuntu bionic-backports InRelease [74.6 kB]                                     
Reading package lists... Done                                                                    
E: The repository 'http://dist.carla.org/carla-0.9.7 bionic Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
N: Skipping acquire of configured file 'main/binary-i386/Packages' as repository 'http://dist.carla.org/carla-0.9.7 all InRelease' doesn't support architecture 'i386'

Installing to Ubuntu 18.04.4 LTS.

Thanks

himanshugoswami, 2020-03-16:
It seems you tried 
**sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys DB53A429E64554FC &&
sudo add-apt-repository "deb [trusted=yes] http://dist.carla.org/carla-0.9.7/ bionic main"**

Before using new installation commands
**sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys DB53A429E64554FC &&
sudo add-apt-repository "deb [trusted=yes] http://dist.carla.org/carla-0.9.7/ all main"**

Which is causing this error: 
**Err:8 http://dist.carla.org/carla-0.9.7 bionic Release
404 Not Found [IP: 34.227.255.250 80]**

Please remove the line **deb http://dist.carla.org/carla-0.9.7/ bionic main** from your /etc/apt/sources.list  file and try again. This should solve your problem.

togaen, 2020-03-16:
@himanshugoswami I was also having this problem and can confirm that the updated commands work. But the documentation needs to be updated: the blog post and documentation still list the old commands. That means anyone who tries installing this according to the docs or the blog is going to run into this error. See:

http://carla.org/2020/03/09/release-0.9.8/

https://carla.readthedocs.io/en/latest/start_quickstart/#downloading-carla

himanshugoswami, 2020-03-16:
@togaen We are working on it. Soon it will be updated. Thanks for the comment.

egreglee, 2020-03-16:
I fell into exactly the hole that @togaen describes.
Thank you, all.

himanshugoswami, 2020-03-20:
Documents are updated.

closed 2020-03-20 by himanshugoswami

mmezl, 2023-12-19:
Hi, I am trying to install Carla using apt-get in Ubuntu 20.04 but probably get no response from> http://dist.carla.org/carla-0.9.7. 
Is it possible the server is not working?

CMakey, 2024-01-02:
> 嗨，我正在尝试在 Ubuntu 20.04 中使用 apt-get 安装 Carla，但可能没有得到任何回应>http://dist.carla.org/carla-0.9.7。服务器有没有可能不工作？
 meet the same error